enesbcs/rpieasy #193: Taskvalueset command has no effect on RunOS command task.

Budman1758, 2021-01-02:
As the title says I have a task setup with the RunOS command and the taskvalueset command does not seem to have any effect. Log shows that the command is sent but the state in the RunOS task does not change. I use a monitored GPIO to determine what state the task is to be in.

On daynighttoggle#State do
if [daynighttoggle#State]=1
taskvalueset,14,1
else
taskvalueset,14,0
endif
endon

That is the rule I'm using.  daynighttoggle is the monitored GPIO.

As an idea for the RunOS command methinks it would be real handy if there was the ability to assign a GPIO pin for it to monitor built into the RunOS plugin. This would simplify things a lot from a rules standpoint. Seperate issue as a feature request maybe?

Let me know what other infos you may need.  Thanks for all you excellent hard work on this software!!

enesbcs, 2021-01-02:
Thanks! Taskvalueset has always waits 3 parameters. Could you try this code?
```
On daynighttoggle#State do
 if [daynighttoggle#State]=1
  taskvalueset,14,1,1
 else
  taskvalueset,14,1,0
 endif
endon
```

Budman1758, 2021-01-02:
Well....  I probably new that back when I was young and my memory was fully functional.  :>))

Thanks for the easy fix!!

enesbcs, 2021-01-02:
Closed as resolved

closed 2021-01-02 by enesbcs