enesbcs/rpieasy #194: Feature request. Extend the Framed OLED plugin.

Budman1758, 2021-01-02:
The Framed OLED plugin in ESPEasy has a couple features that would be nice to have in the RPIEasy version.

Is it possible to add the "display button" and the ability to step thru the frames with it?  Another thing that would be cool is a longpress option to turn the display on or off with the same button.  Both of these would be pretty cool. I realize that the on/off function is available from rules.

Happy New Year!!

enesbcs, 2021-01-02:
I've never used the integrated button function in ESPEasy OLED plugin.
How is it works?
Short press steps next page and long press changes on/off? How long is a long press?

Budman1758, 2021-01-02:
The ESPEasy version does not have the longpress function. I was just extending the idea a bit.  :>)

The step thru frames just uses a momentary pushbutton to work.

enesbcs, 2021-01-02:
Display Button option added at commit https://github.com/enesbcs/rpieasy/commit/9c6ccd5ab4d5091111170fa8e5a465b7727a3e7a

enesbcs, 2021-01-04:
Closed as implemented. Feel free to reopen in case of related issue.

closed 2021-01-04 by enesbcs