enesbcs/rpieasy #197: _P029_DomoOutput  invert logic

Epik77, 2021-01-22:
Hi,
I have to connect a 2 channel 5 volt relay on my rpi
I'm using the Output - Output Helper plugin by connecting MQTT to domoticz.
I need to reverse the logic of the gpio port, can you put a flags in the plugins?

thank's :-)

enesbcs, 2021-01-22:
Its a bit more complicated than just adding a flag, but possible. RPIEasy handles a bit differently the gpio and OutputHelper, if you tried them earlier, you may have observed that if you change the GPIO with "gpio,pin,value" command directly, the OutputHelper value stays in sync with the real value of its assigned pin. (As gpio command checks if the pin is included in an OutputHelper task and changes its value directly.)

enesbcs, 2021-01-22:
Added at commit https://github.com/enesbcs/rpieasy/commit/b33a111a65f4e138c6ee88b1672e87147e397fde

Epik77, 2021-01-22:
yes I understand if there are no other solutions I will have to act at the hardware level
https://docs.google.com/file/d/0B5-HND9HJkXWSTQtYlFTZ3VyODA/edit 

do you have any workaraounds to offer me?

Epik77, 2021-01-22:
Sorry,

after Added at commit b33a111

this device works correctly

thank you :-)

enesbcs, 2021-01-23:
Closed as implemented

closed 2021-01-23 by enesbcs