enesbcs/rpieasy #198: Pass system variable into RunOS command

Budman1758, 2021-01-23:
I am experimenting with a dashcam setup for my truck. I am using a USB camera and the object here is to start and stop recording video via a GPIO, rules and the RunOS command. I am trying to pass the system time into the start record command to place the system time as part of the resulting filename.  Here is the command I'm using...

`ffmpeg -f v4l2 -r ntsc -s  640x480 -t 60 -y -i  /dev/video0 /home/pi/Videos/drive%lcltime%.avi`

This results in filename  >  drive%lcltime%.avi  < instead of the desired effect of the system time as part of the file name. I even tried enclosing it in square brackets. Only difference there is the brackets are added to the file name too. 

I don't know if this is a bug or just not implemented. (or possible).  It would be cool if it could work.

ffmpeg has the ability to add a timestamp to the file name for image files and I assumed it would do the same for video files also but I have not been able to make that work. Hence this method.

Thoughts?
As always, Kudos for this great project!

enesbcs, 2021-01-23:
> This results in filename > drive%lcltime%.avi < instead of the desired effect of the system time as part of the file name. I even tried enclosing it in square brackets. Only difference there is the [...]

Interesting idea. It is not yet implemented, but can be done. I am not sure if it is needed by every user or some linux command lines needs '%' for other purposes, how about adding a checkbox to "Enable command line parsing" to RunOS?

Budman1758, 2021-01-23:
> I am not sure if it is needed by every user or some linux command lines needs '%' for other purposes, how about adding a checkbox to "Enable command line parsing" to RunOS?

That sounds fine to me. Not sure about other command usage of '%' . Whatever checks and balances are needed to make it work for everyone is OK by me.

enesbcs, 2021-01-24:
Command line parse option added at commit https://github.com/enesbcs/rpieasy/commit/17489dff1fd7e6f73baf5ca5c7cfff428d4b3eab

Budman1758, 2021-01-25:
Seems to be working OK. At first I was getting an error message using %lcltime% to the mmpeg command. After some head scratching I figured out that because that variable inserts the  -  character between the numbers that was the error. A  -  in the command line is for options and that is the error source.  Using separate multiple variables fixed that issue.  In short, any variable that inserts a  -  character will probable result in errors on any command that uses  -  for option specifiers.

Now if I could figure out how to send a "stop recording" command this would be perfect. If you start a recording from the command line a simple "CTRL-C" will stop it. I don't know how to make that happen using this setup. My current workaround is to make fixed length recordings and use a timer to start a new one each period of time.

enesbcs, 2021-01-25:
> Seems to be working OK. At first I was getting an error message using %lcltime% to the mmpeg command. After some head scratching I figured out that because that variable inserts the - character [...]

As you will see space in variables can also be source of some problems. When you are using variable strings in commands, try to use quotes like "myfilename%lcltime%.avi" this will eliminate most problems.

> Now if I could figure out how to send a "stop recording" command this would be perfect. If you start a recording from the command line a simple "CTRL-C" will stop it. I don't know how to make that [...]

"sudo killall ffmpeg" command will surely stop all ongoing ffmpeg processes.

Budman1758, 2021-01-25:
> "sudo killall ffmpeg" command will surely stop all ongoing ffmpeg processes.

It surely does.  Thanks!!

enesbcs, 2021-01-27:
Thank you for feedbacks i am closing now as implemented. Feel free to reopen on case of further problems.

closed 2021-01-27 by enesbcs