enesbcs/rpieasy #199: Node list on main page disappears after some time

Budman1758, 2021-01-23:
This began happening right after the update from about a week ago. I first noticed it on my Irrigation control unit. After about a couple hours the node list on the main page completely disappears. Looks as if the ESPEasy P2P controller is turned off. If I go the the controller and resubmit the settings the node list reappears...for a while. 

This is happening on the new testing unit I am using to test my dashcam setup also. I had thought it was weirdness on just the one unit but apparently not.

Another strange thing... The dash cam unit shows up on the node list as something completely different than it should. Attached is a screenshot of the page.  Notice the system info area and compare that to the last entry in the node list. They are the same unit.  I have seen this with other units in the past also. Strange....

[image]

enesbcs, 2021-01-23:
> This began happening right after the update from about a week ago. I first noticed it on my Irrigation control unit. After about a couple hours the node list on the main page completely disappears. [...]

I've found just yesterday a similar issue, when somehow a null sized buffer arrived in UDP port and the P2P decoding process freezed, than no more incoming packages can be processed. :( And the main page filters out the units that did not respond in 5 minutes, than the list will be empty and "disappears". Hard to reproduce but some extra checks will be added to C013.

> Another strange thing... The dash cam unit shows up on the node list as something completely different than it should. Attached is a screenshot of the page. Notice the system info area and compare [...]

I've seen it before on an OrangePI installation, but only one unit does this occasionally, and mainly at startup. Also hard to debug&reproduce: it may be some invalid UDP package or the head of the UDP package is missing somehow. P2P desperately needs some CRC data validation method but if we wants to be compatible, structure can not be altered. Maybe i can add some size based extra checks for incoming packets.

TD-er, 2021-01-23:
> I've seen it before on an OrangePI installation, but only one unit does this occasionally, and mainly at startup. Also hard to debug&reproduce: it may be some invalid UDP package or the head of the [...]

Could it also be left-over data in an allocated piece of data that's not being zeroed?
Or is that something you don't have to care about in Python?
Maybe we should add a checksum field in the p2p messages (we already have a "build" field which we also can use as some kind of version of the message format)
This may help to prevent strange data to be processed.

enesbcs, 2021-01-23:
> Could it also be left-over data in an allocated piece of data that's not being zeroed?

The first command in the loop is a clear() which zeroes out input buffers.

According the picture Unit "73" which stands for "I" in RPIEasytest shifted into the position of the unit field, which has to be "200" instead. Build number 24901 is simply "Ea" from the name and only the rest of the name "sytest" appeared in the name column. There has to be some error in the MAC and or IP address field when info message assembled at the first time.
Nodetype is invalid as i see, maybe i can filter out invalid nodetypes, and drop invalid packages.

> Maybe we should add a checksum field in the p2p messages (we already have a "build" field which we also can use as some kind of version of the message format)

I think the CRC would be an awesome update for the next version (v1) of p2p messages. For the current v0 i do not recommend to alter existing structs as it may breaks compatibility with older systems.

TD-er, 2021-01-23:
That's also why I mentioned the "build nr" as somewhat of a version nr of the message.
For example < 20111 is old format, not using CRC.

enesbcs, 2021-01-23:
> That's also why I mentioned the "build nr" as somewhat of a version nr of the message.
> For example < 20111 is old format, not using CRC.

I see, this may be a good solution. However RPIEasy versioning (based on day of the year) is a bit different than ESPEasy version numbers.

TD-er, 2021-01-23:
There is also a byte signaling whether it comes from ESP8266/ESP32 :)

enesbcs, 2021-01-23:
I think i found the source of the problem. When the network interface is not available, an empty string was provided as IP address, instead of '0.0.0.0'. Will fix ASAP.

enesbcs, 2021-01-23:
In commit https://github.com/enesbcs/rpieasy/commit/608f54396fd8b903d2ed4b152a8680c3b9de9610 :
Empty buffer discarding, and invalid nodetype discarding added to receiver side.
On sender side, invalid IP address now converted to 0.0.0.0 to make sure buffer size will be okay. IP address will be refreshed when the next info packet arrives.

Budman1758, 2021-01-23:
Updated both units. Already seems to have fixed the strange info issue. Time will tell on the disappearing part...

Budman1758, 2021-01-24:
Looks like you squashed that bug. Both of them.  :>)

enesbcs, 2021-01-24:
Good news! Shall we close this issue?

closed 2021-01-24 by Budman1758