enesbcs/rpieasy #200: GPS Position plugin. No Longitude or Latitude data.

Budman1758, 2021-01-26:
Experimenting with the GPS position plugin. I have the Adafruit hat in this [link](https://learn.adafruit.com/adafruit-ultimate-gps-hat-for-raspberry-pi).  The hat is spitting out nema sentences as it should. Like these...

`$GPGSA,A,3,19,28,13,14,15,17,24,30,,,,,1.46,1.15,0.90*0F
$GPGSV,3,1,12,17,87,028,21,19,65,218,22,28,54,036,22,14,49,050,15*7F
$GPGSV,3,2,12,13,44,246,24,30,43,133,20,15,24,287,19,01,20,052,14*7E
$GPGSV,3,3,12,06,19,166,15,24,13,315,15,07,10,135,,21,05,036,*74
$GPRMC,004317.000,A,3358.5144,N,11718.6535,W,0.43,319.80,260121,,,A*7D
$GPVTG,319.80,T,,M,0.43,N,0.79,K,A*37
$GPGGA,004318.000,3358.5145,N,11718.6536,W,1,08,1.15,355.2,M,-32.9,M,,*59
$GPGSA,A,3,19,28,13,14,15,17,24,30,,,,,1.46,1.15,0.90*0F
$GPRMC,004318.000,A,3358.5145,N,11718.6536,W,0.44,311.53,260121,,,A*71
$GPVTG,311.53,T,,M,0.44,N,0.82,K,A*32
$GPGGA,004319.000,3358.5146,N,11718.6537,W,1,08,1.15,355.2,M,-32.9,M,,*5A
$GPGSA,A,3,19,28,13,14,15,17,24,30,,,,,1.46,1.15,0.90*0F
$GPRMC,004319.000,A,3358.5146,N,11718.6537,W,0.43,321.73,260121,,,A*74
$GPVTG,321.73,T,,M,0.43,N,0.80,K,A*36
$GPGGA,004320.000,3358.5147,N,11718.6537,W,1,08,1.15,355.2,M,-32.9,M,,*51
`

However the data for lat and long is nothing but zero in the plugin.  The data for elevation and speed is good and accurate though so it seems to be getting data as it should. I imagine this is probably a parsing error?  I have a different GPS unit right next to it using the same antenna via a splitter that has a pretty darn close fix to where I am. Makes no difference to either setup if using the splitter or not either. Let me know what more information you may need.

Also wondering if this can be used to to get data from gpsd service on the local machine. Haven't tried that, just curious.

TD-er, 2021-01-26:
How did you get these sentences?
The GPRMC sentences all have immediately the GPVTG attached.
So maybe a newline is missing in your received data?
The GPGGA sentence does also have the position info, but if only the GPRMC is parsed for position, you would not get the position updated.

Budman1758, 2021-01-26:
> How did you get these sentences?

That's what the serial port is outputting. Using the instructions on this page is how I made sure the chip is working.
I just copied and pasted the output on the screen to the post above. Using Putty on Windows.
[https://learn.adafruit.com/adafruit-ultimate-gps-hat-for-raspberry-pi/basic-test](url)

The plugin gets its info directly from the serial port.

TD-er, 2021-01-26:
Like I said, it looks like the GPRMC lines in your dump don't have a newline.
If the checksum is computed only if a newline is encountered, you will see a checksum error and likely the line is discarded

The checksum for NMEA data always follows a `*`, so as soon as an asterisk is seen, the next chars are the checksum.
Also a NMEA sentence always starts with a `$`.
This way you don't need to have newlines to process the data.

Budman1758, 2021-01-26:
So how do I change the output?

TD-er, 2021-01-26:
Not sure if you should do it, I just tried to help here as the parser for rpieasy is probably checking for a newline which your GPS output doesn't seem to output here.
So to me it seems like it may be a bug in rpieasy... or at least easier to fix in rpieasy compared to fixing it in the GPS receiver :)

Budman1758, 2021-01-26:
Just for the hell of it I'm gonna install gpsd and gps tools and see what happens there

Budman1758, 2021-01-26:
Here is a bit longer dump. It is exactly as it is in the putty window.

$GPVTG,309.33,T,,M,0.15,N,0.28,K,A*39
$GPGGA,145144.000,3358.5105,N,11718.6422,W,1,04,2.08,368.7,M,-32.9,M,,*5E
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145144.000,A,3358.5105,N,11718.6422,W,0.15,310.93,260121,,,A*77
$GPVTG,310.93,T,,M,0.15,N,0.28,K,A*3B
$GPGGA,145145.000,3358.5109,N,11718.6418,W,1,04,2.08,368.8,M,-32.9,M,,*55
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPGSV,3,1,12,31,67,118,27,21,57,260,16,01,50,295,24,51,49,162,26*70
$GPGSV,3,2,12,22,44,316,19,32,37,049,20,10,21,108,,03,20,308,*7C
$GPGSV,3,3,12,04,14,255,13,25,10,068,,26,08,150,,08,04,229,*7E
$GPRMC,145145.000,A,3358.5109,N,11718.6418,W,0.15,336.92,260121,,,A*76
$GPVTG,336.92,T,,M,0.15,N,0.28,K,A*3E
$GPGGA,145146.000,3358.5111,N,11718.6416,W,1,04,2.08,369.0,M,-32.9,M,,*58
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145146.000,A,3358.5111,N,11718.6416,W,0.16,339.75,260121,,,A*77
$GPVTG,339.75,T,,M,0.16,N,0.30,K,A*32
$GPGGA,145147.000,3358.5115,N,11718.6414,W,1,04,2.08,369.1,M,-32.9,M,,*5E
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145147.000,A,3358.5115,N,11718.6414,W,0.25,338.90,260121,,,A*7A
$GPVTG,338.90,T,,M,0.25,N,0.47,K,A*38
$GPGGA,145148.000,3358.5121,N,11718.6415,W,1,04,2.08,369.6,M,-32.9,M,,*50
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145148.000,A,3358.5121,N,11718.6415,W,0.39,328.70,260121,,,A*71
$GPVTG,328.70,T,,M,0.39,N,0.73,K,A*3D
$GPGGA,145149.000,3358.5123,N,11718.6413,W,1,04,2.08,370.1,M,-32.9,M,,*5A
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145149.000,A,3358.5123,N,11718.6413,W,0.34,336.36,260121,,,A*74
$GPVTG,336.36,T,,M,0.34,N,0.63,K,A*3C
$GPGGA,145150.000,3358.5125,N,11718.6413,W,1,04,2.08,370.5,M,-32.9,M,,*50
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPGSV,3,1,12,31,67,118,27,21,57,260,17,01,50,294,24,51,49,162,26*70
$GPGSV,3,2,12,22,44,316,19,32,37,049,21,10,21,108,,03,20,308,*7D
$GPGSV,3,3,12,04,14,255,14,25,10,068,,26,08,150,,08,04,229,*79
$GPRMC,145150.000,A,3358.5125,N,11718.6413,W,0.31,328.71,260121,,,A*73
$GPVTG,328.71,T,,M,0.31,N,0.57,K,A*32
$GPGGA,145151.000,3358.5125,N,11718.6413,W,1,04,2.08,370.8,M,-32.9,M,,*5C
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145151.000,A,3358.5125,N,11718.6413,W,0.25,325.72,260121,,,A*79
$GPVTG,325.72,T,,M,0.25,N,0.45,K,A*3A
$GPGGA,145152.000,3358.5127,N,11718.6413,W,1,04,2.08,371.2,M,-32.9,M,,*56
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145152.000,A,3358.5127,N,11718.6413,W,0.28,320.37,260121,,,A*71
$GPVTG,320.37,T,,M,0.28,N,0.52,K,A*35
$GPGGA,145153.000,3358.5127,N,11718.6414,W,1,04,2.08,371.4,M,-32.9,M,,*56
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145153.000,A,3358.5127,N,11718.6414,W,0.27,310.12,260121,,,A*7C
$GPVTG,310.12,T,,M,0.27,N,0.50,K,A*3C
$GPGGA,145154.000,3358.5126,N,11718.6413,W,1,04,2.08,371.7,M,-32.9,M,,*54
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145154.000,A,3358.5126,N,11718.6413,W,0.15,305.94,260121,,,A*76
$GPVTG,305.94,T,,M,0.15,N,0.28,K,A*38
$GPGGA,145155.000,3358.5124,N,11718.6413,W,1,04,2.08,371.9,M,-32.9,M,,*59
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPGSV,3,1,12,31,67,118,27,21,57,260,18,01,50,294,24,51,49,162,26*7F
$GPGSV,3,2,12,22,44,316,18,32,37,049,21,10,20,108,,03,20,308,*7D
$GPGSV,3,3,12,04,14,255,13,25,10,068,,26,08,150,,08,04,229,*7E
$GPRMC,145155.000,A,3358.5124,N,11718.6413,W,0.12,290.28,260121,,,A*78
$GPVTG,290.28,T,,M,0.12,N,0.22,K,A*3F
$GPGGA,145156.000,3358.5124,N,11718.6413,W,1,04,2.08,372.0,M,-32.9,M,,*50
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145156.000,A,3358.5124,N,11718.6413,W,0.12,276.85,260121,,,A*74
$GPVTG,276.85,T,,M,0.12,N,0.23,K,A*31
$GPGGA,145157.000,3358.5124,N,11718.6414,W,1,04,2.08,372.0,M,-32.9,M,,*56
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
$GPRMC,145157.000,A,3358.5124,N,11718.6414,W,0.16,285.35,260121,,,A*71
$GPVTG,285.35,T,,M,0.16,N,0.30,K,A*30
$GPGGA,145158.000,3358.5122,N,11718.6414,W,1,04,2.08,372.2,M,-32.9,M,,*5D
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00*0B
[... 226 lines cut ...]
$GPRMC,145247.000,A,3358.5131,N,11718.6420,W,0.23,293.10,260121,,,A*76
$GPVTG,293.10,T,,M,0.23,N,0.43,K,A*32
$GPGGA,145248.000,3358.5131,N,11718.6420,W,1,04,2.10,379.4,M,-32.9,M,,*5E
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.32,2.10,1.00*01
$GPRMC,145248.000,A,3358.5131,N,11718.6420,W,0.22,307.98,260121,,,A*74
$GPVTG,307.98,T,,M,0.22,N,0.40,K,A*3C
$GPGGA,145249.000,3358.5132,N,11718.6420,W,1,04,2.10,379.5,M,-32.9,M,,*5D
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.32,2.10,1.00*01
$GPRMC,145249.000,A,3358.5132,N,11718.6420,W,0.22,316.79,260121,,,A*79
$GPVTG,316.79,T,,M,0.22,N,0.41,K,A*32
$GPGGA,145250.000,3358.5132,N,11718.6421,W,1,04,2.10,379.7,M,-32.9,M,,*56
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.32,2.10,1.00*01
$GPGSV,3,1,12,31,68,117,26,21,57,259,19,01,50,294,22,51,49,162,27*72
$GPGSV,3,2,12,22,45,316,13,32,36,049,23,03,21,308,,10,20,109,*74
$GPGSV,3,3,12,04,14,255,,25,10,067,,26,08,150,,08,04,229,*73
$GPRMC,145250.000,A,3358.5132,N,11718.6421,W,0.30,314.77,260121,,,A*7F
$GPVTG,314.77,T,,M,0.30,N,0.55,K,A*38
$GPGGA,145251.000,3358.5132,N,11718.6423,W,1,04,2.10,379.8,M,-32.9,M,,*5A
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.33,2.10,1.00*00
$GPRMC,145251.000,A,3358.5132,N,11718.6423,W,0.30,310.95,260121,,,A*74
$GPVTG,310.95,T,,M,0.30,N,0.56,K,A*33
$GPGGA,145252.000,3358.5130,N,11718.6423,W,1,04,2.10,380.0,M,-32.9,M,,*55
$GPGSA,A,3,32,31,01,22,,,,,,,,,2.33,2.10,1.00*00
$GPRMC,145252.000,A,3358.5130,N,11718.6423,W,0.25,306.84,260121,,,A*76
$GPVTG,306.84,T,,M,0.25,N,0.46,K,A*31

TD-er, 2021-01-26:
```
$GPRMC,145249.000,A,3358.5132,N,11718.6420,W,0.22,316.79,260121,,,A_79 $GPVTG,316.79,T,,M,0.22,N,0.41,K,A_32
$GPGGA,145250.000,3358.5132,N,11718.6421,W,1,04,2.10,379.7,M,-32.9,M,,_56 $GPGSA,A,3,32,31,01,22,,,,,,,,,2.32,2.10,1.00_01
$GPGSV,3,1,12,31,68,117,26,21,57,259,19,01,50,294,22,51,49,162,27_72 $GPGSV,3,2,12,22,45,316,13,32,36,049,23,03,21,308,,10,20,109,_74 $GPGSV,3,3,12,04,14,255,,25,10,067,,26,08,150,,08,04,229,_73 $GPRMC,145250.000,A,3358.5132,N,11718.6421,W,0.30,314.77,260121,,,A_7F $GPVTG,314.77,T,,M,0.30,N,0.55,K,A_38 $GPGGA,145251.000,3358.5132,N,11718.6423,W,1,04,2.10,379.8,M,-32.9,M,,_5A $GPGSA,A,3,32,31,01,22,,,,,,,,,2.33,2.10,1.00_00 $GPRMC,145251.000,A,3358.5132,N,11718.6423,W,0.30,310.95,260121,,,A_74
$GPVTG,310.95,T,,M,0.30,N,0.56,K,A_33 $GPGGA,145252.000,3358.5130,N,11718.6423,W,1,04,2.10,380.0,M,-32.9,M,,_55 $GPGSA,A,3,32,31,01,22,,,,,,,,,2.33,2.10,1.00_00 $GPRMC,145252.000,A,3358.5130,N,11718.6423,W,0.25,306.84,260121,,,A_76
$GPVTG,306.84,T,,M,0.25,N,0.46,K,A*31
```
Yep missing newlines every now and then...

Budman1758, 2021-01-26:
How are you getting that output?

Also, running RPIEasy from command line I get this message after enabling the gps plugin.

`07:11:24: Try to init serial /dev/serial0 speed 9600
07:11:24: Search for GPS...
can only concatenate str (not "NoneType") to str
`

TD-er, 2021-01-26:
> How are you getting that output?

Wrapping it in "code blocks"
You already wrapped your log in an inline code block using a single backtick 
To create a code block, place 3 backticks on a single line before and after it.

> 
> Also, running RPIEasy from command line I get this message after enabling the gps plugin.
> 
> `07:11:24: Try to init serial /dev/serial0 speed 9600 07:11:24: Search for GPS... can only concatenate str (not "NoneType") to str `

That's somethig for @enesbcs  to look at :)

Budman1758, 2021-01-26:
Below is the output from the "cpgs" program. Lat, long and altitude are dead on to my position.
Speed is a bit off.  Not moving at all.  :>)

lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqklqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
x    Time:       2021-01-26T15:28:20.000Z   xxPRN:   Elev:  Azim:  SNR:  Used: x
x    Latitude:    33.97514833 N             xx   1    53    266    27      Y   x
x    Longitude:  117.31074166 W             xx   3    34    316    17      Y   x
x    Altitude:   363.300 m                  xx  21    49    233    20      Y   x
x    Speed:      0.04 kph                   xx  22    59    322    28      Y   x
x    Heading:    240.6 deg (true)           xx  26    19    139    15      Y   x
x    Climb:      0.00 m/min                 xx  31    66    070    28      Y   x
x    Status:     3D DIFF FIX (45 secs)      xx  32    25    060    16      Y   x
x    Longitude Err:   +/- 2 m               xx   4    24    268    19      N   x
x    Latitude Err:    +/- 3 m               xx  10    09    120    00      N   x
x    Altitude Err:    +/- 5 m               xx  16    01    164    00      N   x
x    Course Err:      n/a                   xx  25    10    052    00      N   x
x    Speed Err:       +/- 23 kph            xx 138    49    162    33      N   x
x    Time offset:     0.835                 xx                                 x
x    Grid Square:     DM13ix                xx                                 x
mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqjmqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

Budman1758, 2021-01-26:
> You already wrapped your log in an inline code block using a single backtick

All I did was copy and paste the screen from putty.

TD-er, 2021-01-26:
In the edit window you can add the 3 backticks wrapping the part you like to have in a code view markup.

N.B. the backtick is the single quote char left of the `1` on your keyboard.

```
This is a code block
multi line
layout
```

You can even add syntax highlighting like this:
```c++
void setup() {
  bla();
}
```
[image]

enesbcs, 2021-01-26:
> > Also, running RPIEasy from command line I get this message after enabling the gps plugin.
> > `07:11:24: Try to init serial /dev/serial0 speed 9600 07:11:24: Search for GPS... can only concatenate str (not "NoneType") to str `
> 
> That's somethig for @enesbcs to look at :)

Very interesting. NoneType means empty buffer was arrived from serial..

enesbcs, 2021-01-26:
> Experimenting with the GPS position plugin. I have the Adafruit hat in this [link](https://learn.adafruit.com/adafruit-ultimate-gps-hat-for-raspberry-pi). The hat is spitting out nema sentences as it should. Like these...
> 
> `$GPGSA,A,3,19,28,13,14,15,17,24,30,,,,,1.46,1.15,0.90*0F 

I've tried simple serial ATGM336H which worked fine. Maybe there were some parsing errors, i will try to input your datas to the parser.

enesbcs, 2021-01-26:
> Here is a bit longer dump. It is exactly as it is in the putty window.
> 
> $GPVTG,309.33,T,,M,0.15,N,0.28,K,A_39 $GPGGA,145144.000,3358.5105,N,11718.6422,W,1,04,2.08,368.7,M,-32.9,M,,_5E $GPGSA,A,3,32,31,01,22,,,,,,,,,2.31,2.08,1.00_0B $GPRMC,145144.000,A,3358.5105,N,11718.6422,W,0.15,310.93,260121,,,A_77
> $GPVTG,306.84,T,,M,0.25,N,0.46,K,A*31

Most lines misses asterisk character which shows the position of checksum.

Budman1758, 2021-01-26:
This is what gpsmon is showing.....

[image]

enesbcs, 2021-01-26:
OK, that looks fine!

enesbcs, 2021-01-26:
> This is what gpsmon is showing.....

I hope commit https://github.com/enesbcs/rpieasy/commit/e1e7efb5f5c875847bb4c686b4e94369d9d384df will solve this.
Could you try the new RPIEasy version?

Budman1758, 2021-01-26:
Looks like you squashed another bug!
Latitude and longitude now displayed correctly.
Speed is displayed as  "none".  Is that as intended?  I would have assumed  "0.0"

enesbcs, 2021-01-26:
> Speed is displayed as "none". Is that as intended? I would have assumed "0.0"

Well None is the default which should be overriden by the first VTG line with valid CRC.

Budman1758, 2021-01-26:
Well... Technically that's not "my test line" all I did was plug the gps hat onto the Pi and turn it on. 
Everything it puts out is its doing, not mine.  :>)

enesbcs, 2021-01-26:
Okay than the module spits the line. :)
I've changed the defaults at commit https://github.com/enesbcs/rpieasy/commit/398c55666f9f2aca2228aba42e73371ebc0035bc

Budman1758, 2021-01-26:
Thats better. :>))

enesbcs, 2021-01-27:
Raspberry's adjusts clock based on NTP servers, and are also has fine linux kernel support for RTC chips.
But GPS can also provide date and time (GPZDA), i am thinking about implementing some time sync capability into this plugin. The only question is the way it should be seen on the user interface (or simple command?). 

@TD-er : are there any plan to support time sync function in espeasy p082 plugin?

Budman1758, 2021-01-27:
You could make an option in the settings where "use ntp" is for "use gps" As for display purpose does it matter what the source of the time is when making use of it? For instance, using time values in rules and in displays the time source itself probably isn't all that important. Or ??

Can the time be added in the plugin values? I think I saw somewhere that 4 values per plugin was some kind of hard limit. Not sure about that. That would be one option if possible.

If I am not mistaken I believe the time from GPS is always GMT? Then you have to convert for your particular time zone?

Just throwing stuff out there...:>)

TD-er, 2021-01-27:
> @TD-er : are there any plan to support time sync function in espeasy p082 plugin?

Already present and in my pending ESP-now PR I extended it to use even more time sources, each with different accuracy rating.

enesbcs, 2021-01-27:
> You could make an option in the settings where "use ntp" is for "use gps" As for display purpose does it matter what the source of the time is when making use of it? For instance, using time values in rules and in displays the time source itself probably isn't all that important. Or ??

Yes, time source is not really relevant for plugins/rules.
"Use ntp" checkbox in "Advanced settings" was global, as ntp is not provided by any plugin but its a core function in RPIEasy.
GPS is a device which may or may not be added to any device slot, which makes a little harder to include in Advanced settings.

> Can the time be added in the plugin values? I think I saw somewhere that 4 values per plugin was some kind of hard limit. Not sure about that. That would be one option if possible.

In theory VARS_PER_TASK in rpieGlobals.py can be modified to a larger number, but it is untested. :) And may break P2P controller.
What do you plan to do with the time on plugin values? A timestamp already exists for every device (sensors only) which contains the last time it collected/reported data. It can be seen on the webGUI when Logging setted to DEBUGMORE.

> If I am not mistaken I believe the time from GPS is always GMT? Then you have to convert for your particular time zone?

Yes it is in GMT, local conversion can be done similarly as when i last played Sunrise/Sunset and Local time. :)

enesbcs, 2021-01-27:
> > @TD-er : are there any plan to support time sync function in espeasy p082 plugin?
> 
> Already present and in my pending ESP-now PR I extended it to use even more time sources, each with different accuracy rating.
> Time sync from GPS was present from the first build that included the GPS plugin.

Thanks i've find it in the source:
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P082_GPS.ino

As i understand P082 will always try to update time, when GPS signal fixed, and the user has no option to disable this?

TD-er, 2021-01-27:
> As i understand P082 will always try to update time, when GPS signal fixed, and the user has no option to disable this?

Yep, as the time given by the GPS is far more superior with regards to drift compared to NTP as that's only done once an hour (when connected).
However there is a big BUT here...
You really should check the data sanity received from the GPS, as I've seen clearly impossible data even though the checksum was correct.
Just as if there was a single byte missing somewhere before it was sent from the GPS and the checkum computed while sending it.
For example a date with 1 byte short, or a GPS position with a missing dot or a missing comma so you're looking at the wrong parameter.

Budman1758, 2021-01-27:
> What do you plan to do with the time on plugin values? A timestamp already exists for every device (sensors only) which contains the last time it collected/reported data. It can be seen on the webGUI when Logging setted to DEBUGMORE.

I don't really have a use case for that. I was more or less just talking about it. No big deal to me... :>)

enesbcs, 2021-03-10:
Original issue solved, further improvements discussed in https://github.com/enesbcs/rpieasy/issues/222

closed 2021-03-10 by enesbcs