enesbcs/rpieasy #201: RPI 4 - Hardware type not shown and unknown GPIO

gary188, 2021-01-27:
Hi,

I have RPIEasy running fine on a 3B+ however on my RPI4 in system info the hardware type is not listed, there are also no autostart options on the hardware page and the GPIO gives the message "This hardware has unknown GPIO".
Looking back through the closed issues a similar issue was raised and addressed to include RPI4 hardware however I still have this problem. Can you assist with this please?

Budman1758, 2021-01-27:
Are you running an older version of the program?

gary188, 2021-01-27:
I"m running 2.1.026, I checked the linux_os.py file to make sure my hardware was listed and the RPI4 is in there.

Budman1758, 2021-01-27:
Interesting. I have 2 PI4's running right now and that all shows and works as it should. Only other thing I can think of is are you running the program as root user? You need that to perform most configuration.

gary188, 2021-01-27:
I have noticed the following errors in the log;

01:44:39: !'hwports' object has no attribute 'webform_load'
01:45:00: Event: Clock#Time=Wed,01:45
Failed to get unit file state for rpieasy.service: No such file or directory
01:45:02: !HW page error 'ram'
01:46:00: Event: Clock#Time=Wed,01:46
01:46:21: !'hwports' object has no attribute 'webform_load'

I noticed this after performing a bluetooth scan and looking through the web GUI

gary188, 2021-01-27:
Yes, started the program with
sudo ./RPIEasy.py

Budman1758, 2021-01-27:
Well... Figuring out that stuff is above my pay grade. :)
I'm sure the developer will weigh in.

enesbcs, 2021-01-27:
> I have RPIEasy running fine on a 3B+ however on my RPI4 in system info the hardware type is not listed, there are also no autostart options on the hardware page and the GPIO gives the message "This hardware has unknown GPIO".

Please show me the result of this command, and i will test it on my installation with your HW ID:

cat /proc/cpuinfo

gary188, 2021-01-27:
Good morning!

The output from command is as follows;
pi@raspberrypi4B:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03114
Serial		: 1000000022052671
Model		: Raspberry Pi 4 Model B Rev 1.4

Hope this help!

gary188, 2021-01-27:
Looking at the file linux_os.py and my hardware info I can see an issue, hardware revision b03114 shows an 8gb Pi4, mine is a 2gb model. This may explain the RAM error in the log if it is trying to address memory that is out of range?

enesbcs, 2021-01-27:
> Looking at the file linux_os.py and my hardware info I can see an issue, hardware revision b03114 shows an 8gb Pi4, mine is a 2gb model. This may explain the RAM error in the log if it is trying to address memory that is out of range?

'b03114' does not exists in current RPIEasy version. I will update this file ASAP.

enesbcs, 2021-01-27:
> Hi,
>
> I have RPIEasy running fine on a 3B+ however on my RPI4 in system info the hardware type is not listed, there are also no autostart options on the hardware page and the GPIO gives the message [...]

enesbcs, 2021-01-27:
revid b03114 and c03114 added at commit
https://github.com/enesbcs/rpieasy/commit/63bb606f4c3678fba04e83fcb457361ee029d6d1

gary188, 2021-01-27:
Thank you very much, that's really appreciated

gary188, 2021-01-27:
Just to confirm that works perfectly now, thanks again!

closed 2021-01-27 by gary188