enesbcs/rpieasy #204: Suggestion: provide a systemd service file

sbrl, 2021-02-18:
By running software under a systemd service file, systemd can monitor it and restart it if it crashes. In addition, it can redirect the logs to a file with syslog.

enesbcs, 2021-02-18:
Systemctl service file support already added at v2.1.023.
Refresh your RPIEasy installation and use appropriate checkbox to enable that on Hardware page.

sbrl, 2021-02-18:
Oh nice! Thanks.

> as with screen i can check command output directly anytime

Indeed. There's another way though, with syslog and logrotate. In the systemd service file, you can put

```ini
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=rpieasy
```

...and then a file at `/etc/rsyslog.d/rpieasy.conf`:

```
if $programname == 'rpieasy' then /var/log/rpieasy/rpieasy.log
if $programname == 'rpieasy' then stop
```

...and then another file at `/etc/logrotate.d/rpieasy`:

```
/var/log/rpieasy/*.log {
	daily
	missingok
	rotate 14
	compress
	delaycompress
	notifempty
	create 0640 root adm
	postrotate
		invoke-rc.d rsyslog rotate >/dev/null
	endscript
}
```

Finally, you can check the command output live like so:

```bash
tail -f /var/log/rpieasy/rpieasy.log
```

Budman1758, 2021-02-19:
> Oh nice! Thanks.
> 
> > as with screen i can check command output directly anytime
> 
> Indeed. There's another way though, with syslog and logrotate. In the systemd service file, you can put
> 
> ```ini
> StandardOutput=syslog
> StandardError=syslog
> SyslogIdentifier=rpieasy
> ```
> 
> ...and then a file at `/etc/rsyslog.d/rpieasy.conf`:
> 
> ```
> if $programname == 'rpieasy' then /var/log/rpieasy/rpieasy.log
> if $programname == 'rpieasy' then stop
> ```
> 
> ...and then another file at `/etc/logrotate.d/rpieasy`:
> 
> ```
> /var/log/rpieasy/*.log {
> 	daily
> 	missingok
> 	rotate 14
> 	compress
> 	delaycompress
> 	notifempty
> 	create 0640 root adm
> 	postrotate
> 		invoke-rc.d rsyslog rotate >/dev/null
> 	endscript
> }
> ```
> 
> Finally, you can check the command output live like so:
> 
> ```shell
> tail -f /var/log/rpieasy/rpieasy.log
> ```

Went thru this procedure and am getting this error message....
`pi@raspberrypi:~ $ tail -f /var/log/rpieasy/rpieasy.log
tail: cannot open '/var/log/rpieasy/rpieasy.log' for reading: No such file or directory
tail: no files remaining`

Seems like its not creating the log file.  Any ideas?  Pretty sure I did exactly as above.

enesbcs, 2021-03-07:
> Seems like its not creating the log file. Any ideas? Pretty sure I did exactly as above.

When you are adding these lines to "/etc/systemd/system/rpieasy.service" file:

StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=rpieasy

Than you can access logs by typing: `cat /var/log/syslog | grep rpieasy`

But these log file only refreshing when you are restarting rpieasy service, so it is not very well for debugging or getting know what is happening just now. So i still prefer rc.local way, simple and working as i need that.

sbrl, 2021-03-08:
Oh, sorry for not replying. Yeah, syslog may buffer the log messages, so it's perhaps not useful for development. But it is a good method for production. You may also need to create the subdirectory it logs to as well before it'll accept it.

closed 2021-03-10 by enesbcs