enesbcs/rpieasy #205: IF in IF ... not working !

Amonbofils, 2021-02-22:
Hi guys !

I got an issue really strange. My script is working on espeasy for a year now, and i try to move to rpieasy for more script quantity. But something went wrong :

My script is

>On System#Boot do
   timerSet,1,1
endon`

>On Rules#Timer=1 do
    if [Status#Status_1]=1
        if [heat_Cool#heatCool_1]=0
            if [Speidel1#Speidel1]>[Setpoint_1#SP_HIGH_1]
                gpio,25,1
            endif
        endif
     else
        gpio,25,0         
    endif
   timerSet,1,10
endon

And log is : 

>Parsed condition: if 0==1 False
20:18:33 Parsed condition: if 0.0==0 True
20:18:33 Parsed condition: if 5.3>0 True
20:18:33 CMD: gpio,25,1
20:18:33 BCM25 set to 1
20:18:33 CMD:`

So there is 3 "IF" commands, but the script is still reading commands even if the first condition is false ...

Any idea to solve it ?

Thanks

enesbcs, 2021-02-22:
Interesting. When i last used espeasy it did not supported embedded IF's, so it is not implemented in RPIEasy. Do you know what espeasy version added this feature?

enesbcs, 2021-02-22:
How about this code?

```
On Rules#Timer=1 do
 timerSet,1,10
 if [Status#Status_1]<1
  gpio,25,0
  breakon
 endif 
 if [Status#Status_1]>0 and [heat_Cool#heatCool_1]<1 and [Speidel1#Speidel1]>[Setpoint_1#SP_HIGH_1]
  gpio,25,1
  breakon
 endif
endon
```

Amonbofils, 2021-02-22:
No i don't know when they added it ... sorry

About your code, it's not working eather because i have multiple code after this one for as many condition needed to manage this part. So i can't break the on condition, also i can use the "and" and "or" conditions, but it'll be very difficult because i have multiple possibilities and multiple if / else embedded. i just shown a basic small part of my code ...

TD-er, 2021-02-22:
> Interesting. When i last used espeasy it did not supported embedded IF's, so it is not implemented in RPIEasy. Do you know what espeasy version added this feature?

Quite a while ago.
The last big refactor of the rules code was Sept 15, 2019, and then it already had:
```c++
  bool condition[RULES_IF_MAX_NESTING_LEVEL];
  bool ifBranche[RULES_IF_MAX_NESTING_LEVEL];
```

enesbcs, 2021-02-22:
> Quite a while ago.
> The last big refactor of the rules code was Sept 15, 2019, and then it already had:
> 
> ```c++
>   bool condition[RULES_IF_MAX_NESTING_LEVEL];
>   bool ifBranche[RULES_IF_MAX_NESTING_LEVEL];
> ```

Thanks! I admire your work and programming skills TD-er, but i have to confess honestly, that i mostly using older ESPEasy sources (~ 2018 december) as for some reason i am unable to get such uptime from newer builds. ;)
This is my favourite build:

Unit Number: | 10
-- | --
Uptime: | 123 days 7 hours 56 minutes
-- | --
Build:⋄ | 20105  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_6_3, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2

I know that ESP core and hardware memory limitation matters a lot in terms of stability, but i have no time and energy to test every release. Sometimes i have a feeling that ESPEasy outgrew itself on ESP8266 hardware in size and complexity.

enesbcs, 2021-02-22:
> About your code, it's not working eather because i have multiple code after this one for as many condition needed to manage this part. So i can't break the on condition, also i can use the "and" and [...]

All right Nested if processing added at commit
https://github.com/enesbcs/rpieasy/commit/3c132fa1bd5dc1b88be74a6fef31ee52e3ca7538
According my tests it runs well on your test code, but more testing may needed.

TD-er, 2021-02-22:
@enesbcs 
Stability is quite OK these days.

[image]

Build Time:⋄ | Oct 21 2020 23:09:36
-- | --
Binary Filename:⋄ | ESP_Easy_mega_20201021_hard_SONOFF_POW_4M1M

[image]

Build Time:⋄ | Dec 19 2020 15:29:05
-- | --
Binary Filename:⋄ | ESP_Easy_mega_20201219_hard_SONOFF_POW_4M1M

enesbcs, 2021-02-22:
> @enesbcs
> Stability is quite OK these days.

96 days looks very nice! Eco mode helps in stability?

Amonbofils, 2021-02-22:
thanks for your help!

So i just need to update on my side ?

enesbcs, 2021-02-22:
> So i just need to update on my side ?

Yes, with Tools->System updates->RPIEasy git update

TD-er, 2021-02-22:
> > @enesbcs
> > Stability is quite OK these days.
> 
> 96 days looks very nice! Eco mode helps in stability?

As always... it depends, but mostly it does the opposite.
Eco mode does call `delay()` whenever noting scheduled has to be executed.
This abundance of `delay()` calls do trigger the (undocumented) power reduction modes of the SDK which do result in quite a dramatic power consumption reduction.

As far as I can tell, it does first lower the TX power to +10 dBm (already close to a factor 10 in TX power reduction) and then to +5dBm.
Then it does increase the DTIM, which is the number of beacon intervals it will listen to the AP and actually turn off the WiFi radio inbetween.
This does cut the power down from 80 mA to 25 mA.
So to name this, I looked at this power reduction and called it "ECO" mode as it is nowhere else documented by Espressif.

I found it by accident when creating the scheduler and this "ECO" mode was more like "how to undo this behavior" as this behavior does cause the ESP to not receive ARP packets.
And if the switches/AP etc. no longer know what MAC is behind what port, no packets can be routed to the ESP anymore, causing all kinds of issues with WDT reboots, stalled communication etc.
So that's why I added the Gratuituous ARP option, to send out the answers to questions nobody asked and just shout out "aa:bb:cc:dd:ee:ff has IP 1.2.3.4" so all network appliances just update their MAC tables.

Amonbofils, 2021-02-24:
I tried it and it's working !Thanks

closed 2021-02-24 by enesbcs