enesbcs/rpieasy #206: %eventvalue% not working

Amonbofils, 2021-02-22:
Hi again !

Still on my project, i'm trying to send values from my HMI to my raspberry. Usually i use the HTTP request with event to send value en espeasy like :

`http://192.168.1.41/control?cmd=event,setTemp_1,12.5`

And on my rpi i have :
On setTemp_1 do
    TaskValueSet 5,1,%eventvalue%
    TaskValueSet 5,2,%eventvalue%+0.5
    TaskValueSet 5,3,%eventvalue%+0.1
endon

but with rpieasy, the value recieved show -1, is it normal ?

thanks

enesbcs, 2021-02-22:
Not really.
In old times, "event,setTemp_1=12.5" worked as eventvalue.
Now the browser somehow trim second = sign from the url.

this might work:
http://192.168.1.41/control?cmd=event,setTemp_1%3D12.5

but i will add support for "," sign style.

enesbcs, 2021-02-22:
Integrated Perver webserver trimmed the = sign. Fixed in commit https://github.com/enesbcs/rpieasy/commit/746e8fd176ff9c2b2e7dd8b2ce239e6eb6ccf52b
Also added support for "," style eventvalue usage.

Amonbofils, 2021-02-24:
Working fine now thanks !

closed 2021-02-24 by enesbcs