enesbcs/rpieasy #207: System crash on ds18B20 failure

Amonbofils, 2021-02-24:
Hi, 

After all modifications you made, i implemented my system in RPIEasy, but after a while, one of my ds18b20 stop working, don't know why yet, but since then, i'm not able to access to "device" page, and most of my rule isn't working.

I had to clear all json to have access to device page again, but i have to rebuild everything...

When i tried to start script manually, this was the output:

 sudo bash ./RPIEasy.py
./RPIEasy.py: ligne 8: import : commande introuvable
./RPIEasy.py: ligne 9: import : commande introuvable
./RPIEasy.py: ligne 10: import : commande introuvable
./RPIEasy.py: ligne 11: import : commande introuvable
from: can't read /var/mail/rpieTime
./RPIEasy.py: ligne 13: import : commande introuvable
./RPIEasy.py: ligne 14: import : commande introuvable
./RPIEasy.py: ligne 15: import : commande introuvable
./RPIEasy.py: ligne 16: import : commande introuvable
./RPIEasy.py: ligne 17: import : commande introuvable
./RPIEasy.py: ligne 18: import : commande introuvable
from: can't read /var/mail/datetime
./RPIEasy.py: ligne 20: try: : commande introuvable
./RPIEasy.py: ligne 21: import : commande introuvable
./RPIEasy.py: ligne 22: except : commande introuvable
./RPIEasy.py: ligne 23: pass : commande introuvable
./RPIEasy.py: ligne 27: erreur de syntaxe près du symbole inattendu « ( »
./RPIEasy.py: ligne 27: `def signal_handler(signal, frame):'

Any idée ?

Amonbofils, 2021-02-24:
After multiple tests, i found something:

After clearing all jsons, i tried to import juste task.json, and it failed, same issue. So i tried to modify the json to found where the problem was.

After many tries, i found that it it was plugin P001 on GPIO who was making the soft crashed. I'm able to import all the json if i delete this P001's part.

But if i delete P001, i can't drive GPIO pins...

enesbcs, 2021-02-24:
>When i tried to start script manually, this was the output:

I hope that you disabled autorun and stopped the background instance before tried to start it manually, as two instances of RPIEasy on the same machine will collide and results strange errors...

> sudo bash ./RPIEasy.py
> from: can't read /var/mail/rpieTime
> from: can't read /var/mail/datetime
> Any idée ?

You are trying to run RPIEasy when active directory is not the directory where rpieTime.py is located. Although "datetime" is an internal python library, strange.

Try to run "sudo ./RPIEasy.py" or "sudo python3 RPIEasy.py" or there are a "run.sh" next to the RPIEasy.py which runs RPIEasy continously with an endless loop after you modified the directory in run.sh.

enesbcs, 2021-02-24:
> After many tries, i found that it it was plugin P001 on GPIO who was making the soft crashed. I'm able to import all the json if i delete this P001's part.

Maybe if you show that json i can tell why. File may be corrupted by multiple running instances or unexpected power outage while writing file.

Amonbofils, 2021-02-24:
i try to restart from scratch, but it failed again, i tried to restore backup, but since everything is stucked... Eveytime it come from P001 device. This time, the "Vanne_1" device was non stop going to enabled to disabled. I deleted it and recreate it... but it crashed and since everithing is blocked, no access to webpanel, so i can't restart or clean it.

Is there a way ton uninstall and reinstall RPIEasy to restart from scratch ?

Here is the json file:

`[{"py/object": "_P004_Dallas.Plugin", "valuenames": ["Speidel1", "", "", ""], "enabled": true, "initialized": true, "pluginid": 4, "taskindex": 0, "valuecount": 1, "taskdevicepin": [-1, -1, -1, -1], "taskdeviceport": "", "dtype": 140, "vtype": 1, "ports": "28-0120426e413b", "uart": -1, "i2c": -1, "spi": -1, "spidnum": -1, "inverselogicoption": false, "pininversed": false, "senddataoption": true, "recdataoption": false, "timeroption": true, "timeroptional": false, "remotefeed": false, "feedpublished": false, "formulaoption": true, "formula": ["%value%", "", "", ""], "decimalsonly": false, "decimals": ["1", 1, 1, 1], "interval": 10.0, "_lastdataservetime": 225109, "timer100ms": false, "timer20ms": false, "timer1s": false, "timer2s": false, "taskname": "Speidel1", "taskdevicepluginconfig": ["28-0120426e413b", 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "readinprogress": 0, "writecallback": null, "uservar": ["13.8", 0, 0, 0], "senddataenabled": [true, false, false, false], "controlleridx": [0, -1, -1, -1], "controllercb": [{"py/object": "builtins.method"}, false, false, false]}, {"py/object": "_P004_Dallas.Plugin", "valuenames": ["Speidel2", "", "", ""], "enabled": true, "initialized": true, "pluginid": 4, "taskindex": 1, "valuecount": 1, "taskdevicepin": [-1, -1, -1, -1], "taskdeviceport": "", "dtype": 140, "vtype": 1, "ports": "28-8000002712f0", "uart": -1, "i2c": -1, "spi": -1, "spidnum": -1, "inverselogicoption": false, "pininversed": false, "senddataoption": true, "recdataoption": false, "timeroption": true, "timeroptional": false, "remotefeed": false, "feedpublished": false, "formulaoption": true, "formula": ["", "", "", ""], "decimalsonly": false, "decimals": ["1", 1, 1, 1], "interval": 10.0, "_lastdataservetime": 225145, "timer100ms": false, "timer20ms": false, "timer1s": false, "timer2s": false, "taskname": "Speidel2", "taskdevicepluginconfig": ["28-8000002712f0", 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "readinprogress": 0, "writecallback": null, "uservar": ["18.6", 0, 0, 0], "senddataenabled": [true, false, false, false], "controlleridx": [0, -1, -1, -1], "controllercb": [{"py/object": "builtins.method"}, false, false, false]}, {"py/object": "_P004_Dallas.Plugin", "valuenames": ["Speidel3", "", "", ""], "enabled": true, "initialized": true, "pluginid": 4, "taskindex": 2, "valuecount": 1, "taskdevicepin": [-1, -1, -1, -1], "taskdeviceport": "", "dtype": 140, "vtype": 1, "ports": "28-01204266f579", "uart": -1, "i2c": -1, "spi": -1, "spidnum": -1, "inverselogicoption": false, "pininversed": false, "senddataoption": true, "recdataoption": false, "timeroption": true, "timeroptional": false, "remotefeed": false, "feedpublished": false, "formulaoption": true, "formula": ["", "", "", ""], "decimalsonly": false, "decimals": ["1", 1, 1, 1], "interval": 10.0, "_lastdataservetime": 227106, "timer100ms": false, "timer20ms": false, "timer1s": false, "timer2s": false, "taskname": "Speidel3", "taskdevicepluginconfig": ["28-01204266f579", 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "readinprogress": 0, "writecallback": null, "uservar": ["14.4", 0, 0, 0], "senddataenabled": [true, false, false, false], "controlleridx": [-1, -1, -1, -1], "controllercb": [{"py/object": "builtins.method"}, false, false, false]}, {"py/object": "_P004_Dallas.Plugin", "valuenames": ["Speidel4", "", "", ""], "enabled": true, "initialized": true, "pluginid": 4, "taskindex": 3, "valuecount": 1, "taskdevicepin": [-1, -1, -1, -1], "taskdeviceport": "", "dtype": 140, "vtype": 1, "ports": "28-0120426158f7", "uart": -1, "i2c": -1, "spi": -1, "spidnum": -1, "inverselogicoption": false, "pininversed": false, "senddataoption": true, "recdataoption": false, "timeroption": true, "timeroptional": false, "remotefeed": false, "feedpublished": false, "formulaoption": true, "formula": ["", "", "", ""], "decimalsonly": false, "decimals": ["1", 1, 1, 1], "interval": 10.0, "_lastdataservetime": 227066, "timer100ms": false, "timer20ms": false, "timer1s": false, "timer2s": false, "taskname": "Speidel4", "taskdevicepluginconfig": ["28-0120426158f7", 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], "readinprogress": 0, "writecallback": null, "uservar": ["14.7", 0, 0, 0], "senddataenabled": [true, false, false, false], "controlleridx": [-1, -1, -1, -1], "controllercb": [{"py/object": "builtins.method"}, false, false, false]}, {"py/object": "_P033_Dummy.Plugin", "valuenames": ["SP_FERM_1", "SP_HIGH_1", "SP_LOW_1", "Dummy4"], "enabled": true, "initialized": true, "pluginid": 33, "taskindex": 4, "valuecount": 3, "taskdevicepin": [-1, -1, -1, -1], "taskdeviceport": "", "dtype": 99, "vtype": 6, "ports": 0, "uart": -1, "i2c": -1, "spi": -1, "spidnum": -1, "inverselogicoption": false, "pininversed": false, "senddataoption": true, "recdataoption": true, "timeroption": true, "timeroptional": true, "remotefeed": false, "feedpublished": false, "formulaoption": true, "formula": ["", "", "", ""], "decimalsonly": false, "decimals": ["1", "1", "1", "1"], "interval": 15.0, "_lastdataservetime": 229206, "timer100ms": false, "timer20ms": false, "timer1s": false [... cut ...] , "", ""], "decimalsonly": false, "decimals": [0, 0, 0, 1], "interval": 10.0, "_lastdataservetime": 226443, "timer100ms": true, "timer20ms": false, "timer1s": false, "timer2s": false, "taskname": "Pompe", "taskdevicepluginconfig": [true, 0, 0, 33, 0, 0, 0, 0, 0, 0, 0, 0], "readinprogress": 0, "writecallback": null, "uservar": ["0", 0, 0, 0], "senddataenabled": [true, false, false, false], "controlleridx": [0, -1, -1, -1], "controllercb": [{"py/object": "builtins.method"}, false, false, false]}]`

enesbcs, 2021-02-24:
> i try to restart from scratch, but it failed again, i tried to restore backup, but since everything is stucked... Eveytime it come from P001 device. This time, the "Vanne_1" device was non stop [...]

There are no installer. You can delete or rename existing directory and reboot. Then download a new one from github.

> Here is the json file:

I've inserted this tasks.json into a test RPIEasy instance, and it is working. Seems that json is valid.

Amonbofils, 2021-02-24:
Yes, it's working now with onlu the task.json, when i add the others jsons, it bricks again... I think it can come from the controler, i'll try to readd it manually.

here are the jsons !

controler.json:
`[{"py/object": "_C008_GenHTTP.Controller", "enabled": true, "controllerip": "192.168.1.150", "controllerport": "8121", "controlleruser": "", "controllerpassword": "", "initialized": true, "controllerid": 8, "controllerindex": 0, "usesID": false, "connected": false, "onmsgcallbacksupported": false, "onmsgcallbackfunc": null, "usesAccount": false, "usesPassword": false, "usesMQTT": false, "timer30s": false, "inchannel": "device=%sysname%&taskid=%id%&cmd=%valname%&value=%value%", "templatestr": "device=RPIEasy&taskid=%id%&cmd=%valname%&value=%value%"}]`

pinout.json:
`[{"ID": 0, "BCM": -1, "name": ["None"], "canchange": 2, "altfunc": 0}, {"ID": 1, "BCM": -1, "name": ["3V3"], "canchange": 0, "altfunc": 0}, {"ID": 2, "BCM": -1, "name": ["5V"], "canchange": 0, "altfunc": 0}, {"ID": 3, "BCM": 2, "name": ["GPIO2", "I2C1-SDA"], "canchange": 1, "altfunc": 1, "startupstate": -1, "actualstate": -1}, {"ID": 4, "BCM": -1, "name": ["5V"], "canchange": 0, "altfunc": 0}, {"ID": 5, "BCM": 3, "name": ["GPIO3", "I2C1-SCL"], "canchange": 1, "altfunc": 1, "startupstate": -1, "actualstate": -1}, {"ID": 6, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 7, "BCM": 4, "name": ["GPIO4"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 8, "BCM": 14, "name": ["GPIO14", "UART-TX", "BT-TX"], "canchange": 1, "altfunc": 2, "startupstate": -1, "actualstate": -1}, {"ID": 9, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 10, "BCM": 15, "name": ["GPIO15", "UART-RX", "BT-RX"], "canchange": 1, "altfunc": 2, "startupstate": -1, "actualstate": -1}, {"ID": 11, "BCM": 17, "name": ["GPIO17", "SPI1-CE1"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 12, "BCM": 18, "name": ["GPIO18/PWM0", "SPI1-CE0", "PCM-CLK"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 13, "BCM": 27, "name": ["GPIO27"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 14, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 15, "BCM": 22, "name": ["GPIO22"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 16, "BCM": 23, "name": ["GPIO23"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 17, "BCM": -1, "name": ["3V3"], "canchange": 0, "altfunc": 0}, {"ID": 18, "BCM": 24, "name": ["GPIO24"], "canchange": 1, "altfunc": 0, "startupstate": 8, "actualstate": 8}, {"ID": 19, "BCM": 10, "name": ["GPIO10", "SPI0-MOSI"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 20, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 21, "BCM": 9, "name": ["GPIO9", "SPI0-MISO"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 22, "BCM": 25, "name": ["GPIO25"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 23, "BCM": 11, "name": ["GPIO11", "SPI0-SCLK"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 24, "BCM": 8, "name": ["GPIO8", "SPI0-CE0"], "canchange": 1, "altfunc": 0, "startupstate": 5, "actualstate": 4}, {"ID": 25, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 26, "BCM": 7, "name": ["GPIO7", "SPI0-CE1"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 27, "BCM": 0, "name": ["ID_SD"], "canchange": 0, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 28, "BCM": 1, "name": ["ID_SC"], "canchange": 0, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 29, "BCM": 5, "name": ["GPIO5"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 30, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 31, "BCM": 6, "name": ["GPIO6"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 32, "BCM": 12, "name": ["GPIO12/PWM0"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 33, "BCM": 13, "name": ["GPIO13/PWM1"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 34, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 35, "BCM": 19, "name": ["GPIO19/PWM1", "SPI1-MISO", "PCM-FS"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 36, "BCM": 16, "name": ["GPIO16", "SPI1-CE2"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 37, "BCM": 26, "name": ["GPIO26"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 38, "BCM": 20, "name": ["GPIO20", "SPI1-MOSI", "PCM-DIN"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}, {"ID": 39, "BCM": -1, "name": ["GND"], "canchange": 0, "altfunc": 0}, {"ID": 40, "BCM": 21, "name": ["GPIO21", "SPI1-SCLK", "PCM-DOUT"], "canchange": 1, "altfunc": 0, "startupstate": -1, "actualstate": -1}]`

Amonbofils, 2021-02-24:
i tried again, with task.json it works, if i add controler manually, everything stop... i'll try again with a long delay between assing task.json and the controler. then i'll unselect all devices to send data to controler. Then i'll add one at the time to send data to contoler until it break.

Amonbofils, 2021-02-24:
Ok, so i tried with only the task.json for 5 minutes, and the program failed again... So it's in the task.json ... i'll try and remove all the P001 devices and see ...

enesbcs, 2021-02-24:
> i tried again, with task.json it works, if i add controler manually, everything stop... i'll try again with a long delay between adding task.json and the controler. then i'll unselect all devices to send data to controler. Then i'll add one at the time to send data to contoler until it break.

what controller?

maybe you can raise intervals on dummy tasks. 1 seconds seems a bit too frequent, effectively spamming mqtt broker if controller is enabled.

Amonbofils, 2021-02-24:
> what controller?
>
> maybe you can raise intervals on dummy tasks. 1 seconds seems a bit too frequent, effectively spamming mqtt broker if controller is enabled.

it's a generic HTTP controler

Amonbofils, 2021-02-24:
is there a way to have access to the logs from ssh ?

enesbcs, 2021-02-24:
> is there a way to have access to the logs from ssh ?

```
sudo su
screen -r
```

enesbcs, 2021-02-24:
FYI: Status_1 has "taskindex": null which is invalid and causes controller to fail.

enesbcs, 2021-02-24:
Added some fix to GenHTTP and startup check if another RPIEasy is running.
https://github.com/enesbcs/rpieasy/commit/6995562ee3027d212adc53c0664b3470a5b85786

Amonbofils, 2021-02-24:
I modified the taskindex, and i changed senddataenabled to false everywhere.

but still crashing after a moment ... 

i'll update to last commit

enesbcs, 2021-02-24:
GPIO8 which you try to use belongs to SPI0, did you disable SPI0?

Amonbofils, 2021-02-24:
Yes, it's disabled. I updated and it seems to work now without controler at the moment, i'll wait few minutes before adding the controler and i'll be back!

Amonbofils, 2021-02-25:
I tried everything ... Nothing went good!

I tried also to create again the devices, one by one, and i tried some restarts, like every 2 devices, and boom... broken!

After reboot one first device goes to "disable" state, and no way to edit it without freezing the system. I'll try to create it again only using "quad" dummy devices...

On your side, everything works ?

enesbcs, 2021-02-25:
> On your side, everything works ?

I've added your controllers.json, pinout.json and tasks.json, and all 16 tasks are working. I've modified the ip address to my http server, to get some answer at least 404.

My console logs:
```
Program exiting gracefully
Starting RPIEasy 2.1.055
Init hardware...
13:00:26: Pi Zero W 40 pins
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
13:00:26: Event: Vanne_1#Vanne_1=0
13:00:26: Registering 10/sec timer as asked
13:00:26: Event: Vanne_3#Vanne_3=0
13:00:26: Registering 10/sec timer as asked
13:00:26: Event: Vanne_4#Vanne_4=0
13:00:26: Registering 10/sec timer as asked
13:00:26: Event: Pompe#Pompe=0
13:00:26: Registering 10/sec timer as asked
Load notifiers from file
13:00:26: Event: System#Boot
13:00:26: Webserver starting at port 80
13:00:26: Event: Clock#Time=Thu,13:00
13:00:26: Event: Network#Connected
13:00:35: !Dallas read error!
13:00:35: !Dallas read error!
13:00:35: !Dallas read error!
13:00:36: !Dallas read error!
13:00:36: Setpoint_2#SP_FERM_2=19.0
13:00:36: Setpoint_2#SP_HIGH_2=19.5
13:00:36: Setpoint_2#SP_LOW_2=19.1
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=6&cmd=SP_FERM_2&value=19.0
13:00:36: Setpoint_3#SP_FERM_3=8.0
13:00:36: Setpoint_3#SP_HIGH_3=8.5
13:00:36: Setpoint_3#SP_LOW_3=8.1
13:00:36: Setpoint_4#SP_FERM_4=9.0
13:00:36: Setpoint_4#SP_HIGH_4=9.5
13:00:36: Setpoint_4#SP_LOW_4=9.1
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=8&cmd=SP_FERM_4&value=9.0
13:00:36: Status#Status_1=0
13:00:36: Status#Status_2=1
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=7&cmd=SP_FERM_3&value=8.0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=6&cmd=SP_HIGH_2&value=19.5
13:00:36: temporary#request_1=0
13:00:36: temporary#request_2=0
13:00:36: Status#Status_3=0
13:00:36: Status#Status_4=0
13:00:36: temporary#request_3=0
13:00:36: temporary#request_4=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=6&cmd=SP_LOW_2&value=19.1
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=7&cmd=SP_HIGH_3&value=8.5
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=16&cmd=Pompe&value=0
13:00:36: Event: Vanne_1#Vanne_1=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=8&cmd=SP_HIGH_4&value=9.5
13:00:36: Event: Vanne_3#Vanne_3=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=14&cmd=Vanne_3&value=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=12&cmd=Vanne_1&value=0
13:00:36: Event: Vanne_4#Vanne_4=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=15&cmd=Vanne_4&value=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=7&cmd=SP_LOW_3&value=8.1
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=8&cmd=SP_LOW_4&value=9.1
13:00:36: Event: Pompe#Pompe=0
13:00:36: http://192.168.2.100:8081/device=pizero&taskid=16&cmd=Pompe&value=0
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
13:00:36: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
```

It seems working without problems, well except the controller is unreachable and i did not connected anything to Raspberry. :)

Although i would use P029 Output device instead P001 Input device, but it should also work. Not so elegant, but works. Isnt it possible that you are done some kind of endless loop with the controller? Perhaps reporting off state to the controller and the controller sending back off command, than reporting also triggered immediately and so on..

Please provide console logs in case of errors.

Amonbofils, 2021-03-04:
I restart everything, and now it seems to work, i'm afraid about rebooting and crashing again..; will see!

Thanks