enesbcs/rpieasy #208: P502_PyGameSnd disable loop

Epik77, 2021-02-26:
Hi,
I need to run preset voice commands the plugins in question would be fine but they run continuously ...
Is it possible to disable the loop or are there any other plugins that I can integrate for this purpose?
Thanks for the support

enesbcs, 2021-02-26:
P503 plugin play sounds once.
But i can add a parameter to specifiy loop count to P502 if it is necessary for your use case.

I am using P502 for continous alarms, and P503 for single welcome sound currently.

Epik77, 2021-02-26:
Yes thanks,
you solve a big problem for me,
I have to upload several mp3s and I would go to occupy many tasks 
thanks :-)

enesbcs, 2021-02-26:
Enhancement: add Loop count parameter to P502

enesbcs, 2021-02-26:
Support added at commit https://github.com/enesbcs/rpieasy/commit/07ae76219f4e73f2b55a48bcf14fb084f7d6953a

Epik77, 2021-02-27:
thank you very much
it works properly :-)

closed 2021-02-27 by Epik77