enesbcs/rpieasy #209: Feature Request: Assign a GPIO to the RunOS command

Budman1758, 2021-03-01:
As the title says I'm wondering if inside the plugin a setting for a GPIO pin to toggle the command could be implemented. This would reduce or eliminate the need for rules to toggle the command.

Thoughts?

enesbcs, 2021-03-01:
Interesting. You are planning to run linux console commands with GPIO connected button presses?

Budman1758, 2021-03-01:
> You are planning to run linux console commands with GPIO connected button presses

Actually it will be kind of automatic. My truck computer will have a dash cam that will record most of the time. Controlled by gpio connected to a box that outputs what gear the truck is in. Put truck in drive, camera records. Put in park, recording stops. Along those lines.

enesbcs, 2021-03-01:
i guess i can add some option like "Input trigger condition" and a GPIO selector. However it will add GPIO as a dependency to use this plugin which is not the best choice.

How about referring to another Task (any task by its name, can be a P001) from the RunOs task, and reacting this trigger task  values?

Budman1758, 2021-03-02:
> However it will add GPIO as a dependency to use this plugin which is not the best choice.

Maybe the GPIO can be optional in the plugin then GPIO would only be necessary if wanted.

enesbcs, 2021-03-02:
> > However it will add GPIO as a dependency to use this plugin which is not the best choice.
> 
> Maybe the GPIO can be optional in the plugin then GPIO would only be necessary if wanted.

"Optional" term is unknown yet in the RPIEasy dependency system. May need a complete rewrite.

Budman1758, 2021-03-03:
> May need a complete rewrite.

Like I said earlier, this is not a big deal. I would not go too far out of the way on this.....

enesbcs, 2021-03-03:
I will find a way to add this feature. :)

enesbcs, 2021-03-04:
Added Trigger Value support at commit https://github.com/enesbcs/rpieasy/commit/eb09739e59bfbd59b97aec83cc6fb2c50c28216e
It will check the linked Task 10 times a second if it's state is changed, and executes the proper command.

Budman1758, 2021-03-07:
That seems to work pretty well. Thanks!!

enesbcs, 2021-03-08:
Closed as implemented

closed 2021-03-08 by enesbcs