enesbcs/rpieasy #210: Timer crash after a while

Amonbofils, 2021-03-04:
I use RPIEasy for a moment, and i found something strange.

In my rules, i created timers, each timer is acting like a loop.

On System#Boot do
    timerSet,1,10
    timerSet,2,10
    timerSet,3,10
    timerSet,4,10
endon

On Rules#Timer=1 do
   ....
   ....
   ....
    timerSet,1,10
endon   

but after 2 or 3 days, timer stop working. I have to create an event to restart timers manually. But it's not really great.

I tried to use loopTimerSet, but it seem to not work...

Any idea ?

enesbcs, 2021-03-04:
> but after 2 or 3 days, timer stop working. I have to create an event to restart timers manually. But it's not really great.

I've 5 rpieasy at home and in the last 2 years i havent experienced such an issue.

> I tried to use loopTimerSet, but it seem to not work...

I do not know what looptimerset is doing. 
timerset set the timer for any interval. once.

enesbcs, 2021-03-04:
> On System#Boot do
> timerSet,1,10
> timerSet,2,10
> timerSet,3,10
> timerSet,4,10
> endon
> 
> On Rules#Timer=1 do
> ....
> ....
> ....
> timerSet,1,10
> endon
> 
> Any idea ?

Create and upload a full rule that reproduces this issue.

tonhuisman, 2021-03-04:
> what looptimerset is doing.

`LoopTimerSet,<timer>,<seconds>[,<count>]` and `LoopTimerSet_ms,<timer>,<msec>[,<count>]` are relatively new ESPEasy commands that provide auto-repeating timers, either endless or a limited count of loops. They have either second or millisecond resolution, and 0 stops the timer. Also, the rules#timer event now is passed a second eventvalue with the current count value.

enesbcs, 2021-03-04:
Millisecond resolution is not really precision under linux. 
LoopTimerSet may be added at a later RPIEasy release.
https://github.com/enesbcs/rpieasy/issues/211

TD-er, 2021-03-04:
It also isn't on an ESP, or at least the timer triggers an event which will be processed in the rules.
So it is more like "subsecond" resolution :)

Amonbofils, 2021-03-08:
Sorry for the delay, still not working at the moment, i still had to restart timers manually every 2 days

here is my code:

```
//*********************
// GPIO
//
// Vanne 1 : 25
// Vanne 2 : 8
// Vanne 3 : 18
// Vanne 4 : 23
//
// Pompe 1 : 27
//*********************

//*********************
//
//     INIT
//
// *********************

On System#Boot do
    timerSet,1,10
    timerSet,2,10
    timerSet,3,10
    timerSet,4,10
endon

On restartTimers do
    timerSet,1,10
    timerSet,2,10
    timerSet,3,10
    timerSet,4,10
endon
//*********************
//
//     EVENTS
//
// *********************
//Request events for fermenting tank 1
On setTemp_1 do
    TaskValueSet 5,1,%eventvalue%
    TaskValueSet 14,1,%eventvalue%+0.5
    TaskValueSet 15,1,%eventvalue%+0.1
endon

On startFerm_1 do
    TaskValueSet 6,1,%eventvalue%
endon
On coolHeat_1 do
    TaskValueSet 7,1,%eventvalue%
endon

//Request events for fermenting tank 2
On setTemp_2 do
    TaskValueSet 5,2,%eventvalue%
    TaskValueSet 14,2,%eventvalue%+0.5
    TaskValueSet 15,2,%eventvalue%+0.1

endon

On startFerm_2 do
    TaskValueSet 6,2,%eventvalue%
endon

On coolHeat_2 do
    TaskValueSet 7,2,%eventvalue%
endon

[... 204 lines cut ...]
                timerSet,5,1
            endif
        else
            if [Speidel4#Speidel4]<[SP_HIGH#SP_HIGH_4]
                gpio,23,1
                if [request#request_4]=0
                    TaskValueSet 8,4,1
                    timerSet,6,20
                endif
            endif
            if [Speidel4#Speidel4]>[SP_HIGH#SP_HIGH_4]
                gpio,23,0
                TaskValueSet 8,4,0
                timerSet,5,1
            endif
        endif
    else
        gpio,23,0
        TaskValueSet 8,4,0
        timerSet,5,1
    endif

    timerSet,4,10
endon
```

enesbcs, 2021-03-08:
> Sorry for the delay, still not working at the moment, i still had to restart timers manually every 2 days. This code was working fine on espeasy for few months (for 2 tanks), i moved to RPI easy to add 2 more tanks, same code. It's working fine for 2 or 3 days and crashed after a while. All other events are still working, datas are still coming from equiments, but timers loops are stopped. By using the event restartTimers, everything start again.
> 
> here is my code:

Thanks for the code, i will test it.
If it is possible please type a space after // comment signs.

enesbcs, 2021-03-08:
@Amonbofils please provide the tasks.json file which matches these Rules.

enesbcs, 2021-03-08:
Maybe a command in the timer On block fails sometimes and terminates the block execution, so next timerset command will never evaluates. Added some fixes in commit https://github.com/enesbcs/rpieasy/commit/5dc83b11258a3cc1f6e1a09f7f68b8ef3dbb255d

Amonbofils, 2021-03-08:
#
[data.zip](https://github.com/enesbcs/rpieasy/files/6104194/data.zip)

Amonbofils, 2021-03-12:
Timers Crashed after 2 days again... I updated the system when you realesed it. I'll change the comments as you indicated and try again.

enesbcs, 2021-03-12:
On my installation:

Uptime: | 3 days 0 hours 19 minutes

and all 4 timers are still running.

Amonbofils, 2021-03-12:
How do you check that timers are running? I had crash again this afternoon...

enesbcs, 2021-03-12:
With the standard way to debug. Through console.

```
sudo su
screen -r
```

On the console i see that timers are stopped and restarted as they has to be.

```
20:52:20: Parsed condition: if 1.0==0 False
20:52:20: Parsed condition: if 19.4<19.1 False
20:52:20: Parsed condition: if 19.4<19.5 True
20:52:20: Parsed condition: if 0.0==0 True
20:52:20: Parsed condition: if 5.1<3.1 False
20:52:20: Parsed condition: if 5.1<3.5 False
20:52:20: Parsed condition: if 19.4>19.5 False
20:52:20: CMD: timerSet,3,10
20:52:20: Timer 3 started with timeout: 10
20:52:20: Parsed condition: if 1.0==0 False
20:52:20: Parsed condition: if 5.1>3.5 True
20:52:20: CMD: timerSet,2,10
20:52:20: Timer 2 started with timeout: 10
20:52:21: Event: Pompe#Pompe=0
20:52:21: http://192.168.2.100:8081/device=RPIEasy&taskid=13&cmd=Pompe&value=0
20:52:21: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: Setpoints#SP_FERM_1=19.0
20:52:22: Setpoints#SP_FERM_2=3.0
20:52:22: Setpoints#SP_FERM_3=19.0
20:52:22: Setpoints#SP_FERM_4=13.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=5&cmd=SP_FERM_1&value=19.0
20:52:22: Status_Tank#Status_1=1.0
20:52:22: Status_Tank#Status_2=1.0
20:52:22: Status_Tank#Status_3=1.0
20:52:22: Status_Tank#Status_4=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=6&cmd=Status_1&value=1.0
20:52:22: heat_Cool#heatCool_1=0.0
20:52:22: heat_Cool#heatCool_2=0.0
20:52:22: heat_Cool#heatCool_3=0.0
20:52:22: heat_Cool#heatCool_4=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=5&cmd=SP_FERM_2&value=3.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=7&cmd=heatCool_1&value=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=6&cmd=Status_2&value=1.0
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=5&cmd=SP_FERM_3&value=19.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=7&cmd=heatCool_2&value=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=6&cmd=Status_3&value=1.0
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=5&cmd=SP_FERM_4&value=13.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=7&cmd=heatCool_3&value=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=6&cmd=Status_4&value=0.0
20:52:22: http://192.168.2.100:8081/device=RPIEasy&taskid=7&cmd=heatCool_4&value=0.0
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:22: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:24: request#request_1=0.0
20:52:24: request#request_2=1.0
20:52:24: request#request_3=0.0
20:52:24: request#request_4=0.0
20:52:24: http://192.168.2.100:8081/device=RPIEasy&taskid=8&cmd=request_1&value=0.0
20:52:24: http://192.168.2.100:8081/device=RPIEasy&taskid=8&cmd=request_2&value=1.0
20:52:24: http://192.168.2.100:8081/device=RPIEasy&taskid=8&cmd=request_3&value=0.0
20:52:24: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:24: http://192.168.2.100:8081/device=RPIEasy&taskid=8&cmd=request_4&value=0.0
20:52:24: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Found
20:52:24: !Controller: 192.168.2.100 connection failed HTTP Error 404: Not Foun
...
20:53:08: Timer 4 stopped
20:53:08: Event: Rules#Timer=4
20:53:08: Parsed condition: if 0.0==1 False
20:53:08: Parsed condition: if 0.0==0 True
20:53:09: Parsed condition: if 14.2>13.5 True
20:53:09: Parsed condition: if 0.0==0 True
20:53:09: Parsed condition: if 14.2<13.1 False
20:53:09: Parsed condition: if 14.2<13.5 False
20:53:09: Parsed condition: if 0.0==0 True
20:53:09: Parsed condition: if 14.2>13.5 True
20:53:09: CMD: gpio,23,0
20:53:09: BCM23 set to 0
20:53:09: CMD: TaskValueSet 8,4,0
20:53:09: Event: request#request_4=0.0
20:53:09: CMD: timerSet,5,1
20:53:09: Timer 5 started with timeout: 1
20:53:09: CMD: timerSet,4,10
20:53:09: Timer 4 started with timeout: 10
20:53:10: Timer 5 stopped
20:53:10: Event: Rules#Timer=5
20:53:10: Parsed condition: if 0.0==0 and 1.0==0 and 0.0==0 and 0.0==0 False
20:53:11: Timer 3 stopped
20:53:11: Event: Rules#Timer=3
20:53:11: Parsed condition: if 1.0==1 True
20:53:11: Timer 2 stopped
20:53:11: Event: Rules#Timer=2
20:53:11: Parsed condition: if 1.0==1 True
20:53:11: Parsed condition: if 0.0==0 True
20:53:11: Parsed condition: if 19.4>19.5 False
20:53:11: Parsed condition: if 0.0==0 True
20:53:11: Parsed condition: if 5.1>3.5 True
20:53:11: CMD: gpio,8,1
20:53:11: Parsed condition: if 0.0==0 True
20:53:11: Parsed condition: if 19.4<19.1 False
20:53:11: BCM8 set to 1
20:53:11: Parsed condition: if 1.0==0 False
20:53:11: Parsed condition: if 19.4<19.5 True
20:53:11: Parsed condition: if 0.0==0 True
20:53:11: Parsed condition: if 19.4>19.5 False
20:53:11: Parsed condition: if 5.1<3.1 False
20:53:11: Parsed condition: if 5.1<3.5 False
```

At least i see Timer2,3,4 and sometimes 5. But it indeed i do not see Timer1, althogh it scrolling fast enough. :)

I guess i will make a menu where timer statuses will be shown.

enesbcs, 2021-03-12:
After commit https://github.com/enesbcs/rpieasy/commit/049fc78b85569dae79aed2f72fa03c9b1d008af2 a hidden service menu is available. 
Type http://your_rpieasy_ip/timers and timer statuses will be displayed

Amonbofils, 2021-03-12:
Awesome! I'll check it! Did you add the loop Timer yet?

Amonbofils, 2021-03-13:
Since updated...

[image]

Amonbofils, 2021-03-13:
Is it possible the error came from a reading error on ds18b20? Value can be "out of range"? On espeasy, you can choose what value is chosen if there is a reading error, what about you? What value is used?

enesbcs, 2021-03-13:
> Awesome! I'll check it! Did you add the loop Timer yet?

Not. As i labeled at looptimer issue, it may take much more time. Basic structures are ready without working code.

enesbcs, 2021-03-13:
> On espeasy, you can choose what value is chosen if there is a reading error, what about you? What value is used?

P004 plugin returns 0 in case of error.

enesbcs, 2021-03-13:
> [image]

Now you can narrow debugging as the line is inside the Timer1 event handler.
You can check it by excluding one line a time and check if it is still happening, as it is not happening in my test environment. I guess either a taskvalueset or a gpio command results a strange error somehow.

```
On Rules#Timer=1 do
//Tank 1
    if [Status_Tank#Status_1]=1
        if [heat_Cool#heatCool_1]=0
            if [Speidel1#Speidel1]>[SP_HIGH#SP_HIGH_1]
                gpio,25,1
                if [request#request_1]=0
                    TaskValueSet 8,1,1
                    timerSet,6,20
                endif

            endif
            if [Speidel1#Speidel1]<[SP_LOW#SP_LOW_1]
                gpio,25,0
                TaskValueSet 8,1,0
                timerSet,5,1
            endif
        else
            if [Speidel1#Speidel1]<[SP_HIGH#SP_HIGH_1]
                gpio,25,1
                if [request#request_1]=0
                    TaskValueSet 8,1,1
                    timerSet,6,20
                endif
            endif
            if [Speidel1#Speidel1]>[SP_HIGH#SP_HIGH_1]
                gpio,25,0
                TaskValueSet 8,1,0
                timerSet,5,1
            endif
        endif
    else
        gpio,25,0
        TaskValueSet 8,1,0
        timerSet,5,1
    endif

    timerSet,1,1
endon
```

enesbcs, 2021-03-13:
I've added further exception checking in case the list index error came from "IF condition nesting level" feature, added 19 days ago.
https://github.com/enesbcs/rpieasy/commit/d0b8994ed50629b80298582339023b108baa18d6

enesbcs, 2021-03-16:
v2.1.072 timers running since 1 days 15 hours 26 minutes on my test environment, i think it will be solved now.

enesbcs, 2021-03-17:
uptime: 2 days 13 hours 4 minutes
timers OK

Closed now, feel free to reopen if further debug informations are available and the error occurs againg.

Amonbofils, 2021-04-17:
Hi again, crash happened again, don't know exactly when and why but it crashed a couple of time a week.

Is it possible to test in a rule the timer value as a watchdog and restart it by script? I saw the page /timers to look on timer statuses. So I think it's possible to check this value. Right?

Thanks

enesbcs, 2021-04-17:
Did you try looptimerset command?

enesbcs, 2021-04-17:
> So I think it's possible to check this value. Right?

Timer has no value. Please be a little more specific.

Amonbofils, 2021-04-17:
On the page timer, you show the time elapsed since the last start of timers. When you look at http://ipadress/timers.

Is it possible to get this value into rules as watchdog?

enesbcs, 2021-04-18:
Yes timer object in RPIEasy has values, like starttime, last start time and timeout. I guess you wants to see actual time - last start time in some form.. like a global variable like %timer1laststart% or something like that which can be examined in another event.

But the best resolution would be to check what lines exactly causes the timers to stop.

Amonbofils, 2021-04-18:
Sure it'd be nicer to know why, but it occurred so randomly, and without a complete log it will be very difficult to know where and when. It occurred also on one or 2 timers but not in all, and the script is the same for all...

Did you worked on the loop Timer? It could help also if the loop Timer is not dependent on script errors

enesbcs, 2021-04-18:
> Did you worked on the loop Timer? It could help also if the loop Timer is not dependent on script errors

Yes, as i wrote in its issue page, it is implemented 14 days ago.
https://github.com/enesbcs/rpieasy/issues/211

Amonbofils, 2021-04-18:
sorry i didn't see it, shame on me ! I updated my script with loop timer. we'll see, i'll keep you updated.

Thanks

Amonbofils, 2021-04-30:
Hi again !

After a while using loop instead of timer, same thing happen...
[image]

is there a way to restart automatically the loop if it's stopped ?

thanks

enesbcs, 2021-05-01:
LoopTimer auto-watchdog added at commit https://github.com/enesbcs/rpieasy/commit/e7ab79ae2003de270211845afa0d643f6e80d2ac

enesbcs, 2021-05-15:
closed as no further informations received

closed 2021-05-15 by enesbcs