enesbcs/rpieasy #212: Varous smaller issues

clumsy-stefan, 2021-03-05:
Hi, Im actively developing ESPEasy modules and helping the maintainer fixing issues. Currently I hvae roughly 30 ESPEasy nodes running productively with a variety of sensors attached. Now as one of my nodes reached it's limits (in terms of speed and mainly number of possible tasks) I was looking for a different solution and came across this great piece of software! So first, thanks a lot for developing!

Now while migrating my node (a wemos-pro) to a RPI3 I came across a few smaller issues in RPIEasy, which are all not limiting or can be worked around, but in terms of feedback and possible enhancements I compiled a list here. I'm happy to support in developing, but my python knowledge is very limited, so I can mainly test and report.

1. When using rules it seems that the possibility to test directly in the `on` can't be used. Eg. Multiple rules on the same event with something like `on pump#speed>3 do` and another one testing for `on pump#speed<0 do` wont will not work. Of course the workaround is to have one rule with the event and test for the value with `if` statements.
2.  Using comments within `on` statements with `//` produce an error. I couldn't find a way to add comments within a rule. Seems just to be a cosmetic issue
3. I'm using a lot of PCFGPIO expansions. Now when setting a port with the `pcfgpio` command, the port gets set, but (as propably then it's assumed to be an output port) the new state is not sent to the controller and the associated task for that port is not updated. As soon as I do a `taskRun` of the associated task, the status gets updated and is sent to the controller, so that's an easy workaround for me. Probably to be noted, I use the PCF?s without the interrupt pin but with the regular polling.
4. Also with PCFGPIO's when not using the interrupt pin but the regular polling (and if I'm reading the code right) the polling is only done every 10ms if the send to controller `Interval` is set to `0`, with all other values it's updated every second, which is "good enough". But I'd appreciate if the polling interval would be more frequent also when specifying an `Interval` bigger than `0`. Eg. I'm usually sending every 5min. or so the value to the controller (kind of a heartbeat), but would still like to react really quick on a button-press attached to this PCFGPIO.
5. Just cosmetics: in the log it says `PCFPIO` instead of `PCFGPIO`.
6. Not sure why this one happens, but it seems either json loading/saving or handling of the tasks have issues if two tasks have the same `name`. I'm using this to send different values to the same topic/device of the controller. Unfortunately I did not yet find how to reproduce and it show no error in the log. Only found, that the version of jsonpickle seems to be significant for RSPIEasy.
7. Sometimes I can't `edit` a task anymore. Also not sure when it happens, but if it happens the browser just hangs. No error in the logs. Probably related to 6) but can't reproduce. The only possibility I found is restoring the tasks.json file and restart RPIEasy.
8. P003_Pulse: we had in ESPEasy a longer discussion about the precision of the pulse counter (see [#3499](https://github.com/letscontrolit/ESPEasy/pull/3499)). We decided in the end to keep `milis()` precision in the values, but internally calculate in `micros()` and return a float value. The issue was that roughly starting at 5'000 pulses/sec the `Time` value of the plugin get's rounded too much. Seems to be an easy and non intrusive change though.
9. On various occasions I saw that `taskValuesStAndRun` will not always update the controller and send the values. Easy to work around with a `taskRun`statement afterwards.
10. It seems that the `(pcf)gpio` commands are only enabled when at least one "regular" gpio task is defined. So when only using PCFGPIO's the commands are not available.

I'm sorry for the long list and as said, all of this is not crucial or meant to diminish the software in any way! I really appreciate what has been done and like to thank you for the big work that has been done. Please let me know if I can help, test or report in any way!

Kind regards!

STefan

enesbcs, 2021-03-05:
> Hi, Im actively developing ESPEasy modules and helping the maintainer fixing issues. Currently I hvae roughly 30 ESPEasy nodes running productively with a variety of sensors attached. Now as one of [...]

Thank you for your feedback!

> 1. When using rules it seems that the possibility to test directly in the `on` can't be used. Eg. Multiple rules on the same event with something like `on pump#speed>3 do` and another one testing [...]

When i started to code rule engine in python three years ago, ESPEasy did not handle multiple rules on the same event name. I guess than now it is capable it, so i have to rewrite the rule engine... sigh.
 
>     2. Using comments within `on` statements with `//` produce an error. I couldn't find a way to add comments within a rule. Seems just to be a cosmetic issue

Please place a space before // and it will work on all lines, like: ' //'
As // can be inside http://.. it is the most fool-proof and simplest approach, from my point of view.

> 3. I'm using a lot of PCFGPIO expansions. Now when setting a port with the `pcfgpio` command, the port gets set, but (as propably then it's assumed to be an output port) the new state is not sent to [...]

I guess you are not a Domoticz user. When Domoticz sends a pin change command, and receives the same answer from the same IDX, it will be in an endless loop. Only inputs are reported when i last checked, as in the P029 plugin. Can be added a report option for non-Domoticz users if needed.

> 4. Also with PCFGPIO's when not using the interrupt pin but the regular polling (and if I'm reading the code right) the polling is only done every 10ms if the send to controller `Interval` is set to [...]

If you are setting and interval, you are waiting the pin status at a specific time. When you are set it to 0 it will be report every time, when it is changing.
I don't really understand the point of these question. You can force a timed report on a 0 interval task with a timer and taskrun, if you wish.

>     5. Just cosmetics: in the log it says `PCFPIO` instead of `PCFGPIO`.

You are right, it will be corrected in the next release.

> 6. Not sure why this one happens, but it seems either json loading/saving or handling of the tasks have issues if two tasks have the same `name`. I'm using this to send different values to the same [...]

You are right, using the same names will lead errors. I will add a check to deny adding the same names to tasks.
https://github.com/enesbcs/rpieasy/issues/213
 
>     7. Sometimes I can't `edit` a task anymore. Also not sure when it happens, but if it happens the browser just hangs. No error in the logs. Probably related to 6) but can't reproduce. The only possibility I found is restoring the tasks.json file and restart RPIEasy.

Yes, same names results funny things. Also multiple running RPIEasy instances results very-very strange errors, which is blocked in the latest release.

> 8. P003_Pulse: we had in ESPEasy a longer discussion about the precision of the pulse counter (see [#3499](https://github.com/letscontrolit/ESPEasy/pull/3499)). We decided in the end to keep [...]

Honestly, in RPIEasy all values stored as strings, and converted when needed.
Timer precision under Linux may be almost fine at millisecond level, but unusable at microsecond precision. +/-1000 microsecond deviance = +/- 1 millisecond. It does not make much sense.
 
>     9. On various occasions I saw that `taskValuesStAndRun` will not always update the controller and send the values. Easy to work around with a `taskRun`statement afterwards.

taskrun calls the plugin plugin_read() method, but taskvaluesetandrun sets the value, and force the publish parameter to true. In theory it will send data to controller if senddataoption is enabled for the selected plugin. (senddataoption value depends on the plugin type)

>     10. It seems that the `(pcf)gpio` commands are only enabled when at least one "regular" gpio task is defined. So when only using PCFGPIO's the commands are not available.

It is really interesting, as RPi.GPIO is needed, but there are no other dependency to either P001 nor P029 defined in P019.

>         EDIT2: I guess I found it: if you use something like `if [var#11]` in rules and this variable has not been set yet (since start), above error is raised. It's a bit misleading as the error happens within a `on Rules#Timer=2 do` rule.

Every global var has an initial value as 0. Can you provide a full rule to reproduce that?

>     12. Probably you could add a `.gitignore` with the `files/` and `data/` directories so that it's easier to pull when local changes have been made to the configuration (no need to stash)..

Upstream "files" and "data" directory contains only one file for reservation. But these directories needs to be created on the first pull, otherwise funny things will happen. At least RPIEasy will not start at all. :)

TD-er, 2021-03-05:
> Upstream "files" and "data" directory contains only one file for reservation. But these directories needs to be created on the first pull, otherwise funny things will happen. At least RPIEasy will not start at all. :)

As far as I know, if the files/dirs exist and then you add the folder to the .gitignore, only changes will not be allowed in future commits, thus all existing files/folders are still preserved and thus created if you clone the project.

clumsy-stefan, 2021-03-05:
First, thanks for the quick reply! Really appreciated.

> so i have to rewrite the rule engine... sigh.

There is a quite "easy" workaround, so I would say, that doesn't have to be a first priority, but when there's nothing else left to do ;)

 > Please place a space before // and it will work on all lines, like: ' //'

Perfect! thanks!

> I guess you are not a Domoticz user. When Domoticz sends a pin change command, and receives the same answer from the same IDX, it will be in an endless loop. Only inputs are reported when i last checked, as in the P029 plugin. Can be added a report option for non-Domoticz users if needed.

Nope, I'm using the FHEM controller. The nice thing about the PCF's are, that you can use the same pin as Input and Output simultaniously. Eg. have a relais controlled by a pin and a switch to manually "override" the output locally (in case of issues) by connecting it to VCC or GND and then report the actual state back to the software.

> I don't really understand the point of these question. You can force a timed report on a 0 interval task with a timer and taskrun, if you wish.

Yes, that would be an option. But it's the situation described above. eg. if a change is "missed" (whyever) it's made sure that it's sent back to the controller every `Interval` time. However the main usage for me with the `Interval` setting is kind of a "heartbeat". FHEM outdates readings/devices if no update is received after a certain time and the device is considered "absent". So if no input change is registered within this time no status is sent back to the controller and FHEM thinks it's dead. Besides it's different behaviour than ESPEasy which checks the pins regardless of the `Interval` setting. So Rules and logic have to be adapted and can't be taken 1:1 from ESPEasy nodes (which is not a big issue for me).

> You are right, using the same names will lead errors. I will add a check to deny adding the same names to tasks.

From my point of view it would be nicer to be able to use duplicated task names and make the combination of <taskname>/<valuename> unique. We had the same discussion on ESPEasy (where it's possible but throws a warning). Probably this doesn't make sense for MQTT users, but for FHEM it does ;)

> Timer precision under Linux may be almost fine at millisecond level, but unusable at microsecond precision. +/-1000 microsecond deviance = +/- 1 millisecond. It does not make much sense.

Ok, makes sense.

> In theory it will send data to controller if senddataoption is enabled for the selected plugin.

Hmm... strange, it did not work for me. I'll test again.

> Every global var has an initial value as 0. Can you provide a full rule to reproduce that?

```
on Rules#Timer=2 do
  event,ch9r=1 // Heizung aus
  if [var#11] < 8
    event,ch5d=0 // Lueftung aus
  endif
endon
```
That's where it threw the error on the `if` statement.

> Upstream "files" and "data" directory contains only one file for reservation. But these directories needs to be created on the first pull, otherwise funny things will happen. At least RPIEasy will not start at all. :)

Probably I stated wrongly, as @TD-er mentioned, in a pull the file is still fetched I guess, but when you do a `git status` it doesn't show these added and changed files as not in line with upstream, makes it easier to pull without stashing (I guess) and local changes are not overridden by default. But I'm also not the real specialist there ;)

clumsy-stefan, 2021-03-05:
Another strange thing which I'm not sure where it comes from: after some time the following task:
[image]
stops sending data to the controller. When I manually hit the "run" button it just does nothing. Also when I try to "edit" it just hangs and doesn't open the config page.

I had this effect before, but couldn't yet pin down when or why it happens. After restarting ESPEasy it show the following:
[image]

Only manually changing the `tasks.json` file and restarting solves it...

clumsy-stefan, 2021-03-06:
Another smaller one: I have a rule which triggers an event with `Event,pwm#speed=%eventvalue%` resulting in the following:
````
09:18:56 CMD: pwm,18,77
09:18:56 BCM18 PWM 77% 1000Hz
09:18:56 Event: pwm#speed=77
09:18:56 CMD: SendToHTTP 10.0.0.1,8072,/fhem?cmd=setreading%20raspi_02_ch05%20speed%2077
09:18:56 CMD: TaskRun,3
09:18:56 CMD: TaskRun,4
09:18:56 CMD: Event,pwm#speed=77
09:18:56 Event: vent=pwm#speed=77
09:18:56 CMD: TaskRun,21
09:18:56 Event: ch5#switch=0
```
Somehow the capital "E" gets stripped and the event is not calld. If I change it to `event,pwm#speed=%eventvalue%` with small "e" it works as intended:
```
09:17:12 BCM18 PWM 60% 1000Hz
09:17:12 Event: pwm#speed=60
09:17:12 CMD: SendToHTTP 10.0.0.1,8072,/fhem?cmd=setreading%20raspi_02_ch05%20speed%2060
09:17:12 CMD: TaskRun,3
09:17:12 CMD: TaskRun,4
09:17:12 CMD: event,pwm#speed=60
09:17:12 Event: pwm#speed=60
09:17:12 CMD: SendToHTTP 10.0.0.1,8072,/fhem?cmd=setreading%20raspi_02_ch05%20speed%2060
09:17:12 CMD: TaskRun,3
09:17:12 CMD: TaskRun,4
09:17:12 CMD: TaskRun,21
09:17:12 Event: ch5#switch=0
````
Don't be surprised about the log showing the event twice, as I also have a task that triggers this event (PWM_Helper).

But I think the issue is about capital and small command names.

enesbcs, 2021-03-06:
> > so i have to rewrite the rule engine... sigh.
> 
> There is a quite "easy" workaround, so I would say, that doesn't have to be a first priority, but when there's nothing else left to do ;)

Sorry, but there is nothing "easy" in the RPIEasy rule engine.
Except the if condition parsing which is done by the python eval, which is easy for me.
https://github.com/enesbcs/rpieasy/issues/214

> Nope, I'm using the FHEM controller. The nice thing about the PCF's are, that you can use the same pin as Input and Output simultaniously. Eg. have a relais controlled by a pin and a switch to [...]

https://github.com/enesbcs/rpieasy/issues/215
 
> From my point of view it would be nicer to be able to use duplicated task names and make the combination of `taskname`/`valuename` unique. We had the same discussion on ESPEasy (where it's possible but throws a warning). Probably this doesn't make sense for MQTT users, but for FHEM it does ;)

Same task names will ruin RPIEasy low level functions which returns taskid from taskname... which is used by many places, for example in all controllers which is not IDX based and all inside message routing fallbacks to tasknames, in case the taskid is not known by the sender.
 
> > In theory it will send data to controller if senddataoption is enabled for the selected plugin.
> 
> Hmm... strange, it did not work for me. I'll test again.

Please add some details about the device type and settings, if it is not working.

> ```
> on Rules#Timer=2 do
>   event,ch9r=1 // Heizung aus
>   if [var#11] < 8
>     event,ch5d=0 // Lueftung aus
>   endif
> endon
> ```
> 
> That's where it threw the error on the `if` statement.

I've added some code to your sample, to test it:
https://github.com/enesbcs/rpieasy/issues/216

enesbcs, 2021-03-06:
> Another strange thing which I'm not sure where it comes from: after some time the following task:
> [image]
> stops sending data to the controller. When I manually hit the "run" button it just does nothing. Also when I try to "edit" it just hangs and doesn't open the config page.

What interval did you use to sysinfo?
The "load" option may need 2 seconds at least, this call is (ironically) a little cpu/time intensive.

enesbcs, 2021-03-06:
> Somehow the capital "E" gets stripped and the event is not calld. If I change it to `event,pwm#speed=%eventvalue%` with small "e" it works as intended:

> But I think the issue is about capital and small command names.

https://github.com/enesbcs/rpieasy/issues/217

clumsy-stefan, 2021-03-06:
> Sorry, but there is nothing "easy" in the RPIEasy rule engine.

Misunderstanding, I didn't mean the rule engine is easy..  I meant the rewriting of the rules so it uses one rule per event and then check for the values inside is easy (for the enduser)....

> What interval did you use to sysinfo?

300sec. After restart I could "edit" again, but the "Interval" option was not shown anymore. So I deleted and added the task again and until now it works. As written I found another duplicate task name (ch5), corrected it (made it uniq) and tested it again. until now it seems to work. So all kind of strange issues with duplicate task names. Therefore I completely agree with you, there should be a check to make sure that there are no duplicate task names.

Again, I'm really sorry to report all these things (at once) and of course I'd never expect it to be fixed, if not wanted by the maintainer... It's really jsut what I came around. The Software is really great though and these things are not really "limiting"!

So many thanks for your efforts and help!! Really appreciated!!

enesbcs, 2021-03-06:
> Misunderstanding, I didn't mean the rule engine is easy.. I meant the rewriting of the rules so it uses one rule per event and then check for the values inside is easy (for the enduser)....

Yep, for the endusers, you are right. :)

> Again, I'm really sorry to report all these things (at once) and of course I'd never expect it to be fixed, if not wanted by the maintainer... It's really jsut what I came around. The Software is [...]

Thank you for your feedback! As i am using old ESPEasy binaries, i am not always up-to-date with current features, but i am trying to be compatible.
I've added several github issues based on your reviews, please add unique issues one per case if something new popping up or missed.

enesbcs, 2021-03-10:
Open new issues in case of problems, closing this now.

closed 2021-03-10 by enesbcs