enesbcs/rpieasy #215: P019: add option to report status to the controller even if type is output

enesbcs, 2021-03-06:


clumsy-stefan, 2021-03-06:
Just for testing I added the following statement to P019 after line 92 (before `try:`. This seems to work for my setup, but can't say what happens to MQTT obviously.
```
     self.timer100ms = True
     misc.addLog(rpieGlobals.LOG_LEVEL_DEBUG,"PCF 1/10s timer enabled with update")
```

enesbcs, 2021-03-06:
> Just for testing I added the following statement to P019 after line 92 (before `try:`. This seems to work for my setup, but can't say what happens to MQTT obviously.
> 
> ```
>      self.timer100ms = True
>      misc.addLog(rpieGlobals.LOG_LEVEL_DEBUG,"PCF 1/10s timer enabled with update")
> ```

May work, as handler will only send data if actual state is not equal to the last reported state. (but may add extra cpu usage of course as if you are not using interrupt callbacks, polling requires more resource everytime)
I can add an option to activate 1/10s timer manually to the plugin settings.

clumsy-stefan, 2021-03-06:
Yes an option sure would be nice. If I'm not wrong, I added the counter where the check for interrupt callback is false (and only then).

enesbcs, 2021-11-23:
Found a typo in the plugin, i think that this workaround is not necessary anymore. 
https://github.com/enesbcs/rpieasy/commit/fadb8031612dd6efbce9efdd9c9f97999a9e8514

clumsy-stefan, 2021-11-25:
Seems to be working (without my workaround) now! Thanks a lot!

enesbcs, 2021-11-25:
Thanks for testing, closing for now!

closed 2021-11-25 by enesbcs