enesbcs/rpieasy #216: Global variables initial values may throw error in rule conditions

enesbcs, 2021-03-06:
Test setup:
 ```
 on ch9r do
   taskvalueset,1,1,%eventvalue%
 endon

 on ch5d do
   taskvalueset,1,2,%eventvalue%
 endon

 on Rules#Timer=2 do
   event,ch9r=1
   if [var#11] < 8
     event,ch5d=0
   endif
 endon
 ```

enesbcs, 2021-03-06:
Fixed in commit https://github.com/enesbcs/rpieasy/commit/eae4618e06fc46d00d3c99469d195b8610ab8429

clumsy-stefan, 2021-03-06:
Thanks for the fix, seems to work:
```
22:33:46: Event: Rules#Timer=2
22:33:46: CMD: event,ch9r=1
22:33:46: Event: ch9r=1
22:33:46: CMD: pcfgpio,9,1
22:33:46: PCFGPIO9 set to 1
22:33:46: CMD: TaskRun,25
22:33:46: Event: ch9#switch=1
22:33:46: Sending task 25 data to FHEM at http://10.0.0.1:8383/ESPEasy
22:33:46: Parsed condition: if 0 < 8 True
22:33:46: CMD: event,ch5d=0
```

The check `0 < 8` is executed without throwing an error, even when the variable is not initialised.

enesbcs, 2021-03-07:
Closed as solved

closed 2021-03-07 by enesbcs