enesbcs/rpieasy #218: P026 Load - cpu usage throws unexpected errors occasionally

enesbcs, 2021-03-06:


enesbcs, 2021-03-06:
Extra checks added at commit https://github.com/enesbcs/rpieasy/commit/eae4618e06fc46d00d3c99469d195b8610ab8429
It may help, but further testing needed.

clumsy-stefan, 2021-03-06:
Not sure what/how to test here. I guess the issue I experienced were because of the duplicated tasknames.

clumsy-stefan, 2021-03-06:
Hmm.. Not sure if it's the right issue here, but it happened again just now in the sysinfo plugin, but now I'm sure I don't have and duplicated task names (checked twice). Suddenly the plugin shows:
[image]
and can't be edited anymore!

If I try to run it, I get the following error on the console:
`'NoneType' object is not subscriptable`

The only way I found to recover is removing it directly in the json file and re-add it.

enesbcs, 2021-03-06:
> If I try to run it, I get the following error on the console:
> `'NoneType' object is not subscriptable`

Is it "P026 'NoneType' object is not subscriptable" ?
When a plugin freezes in Edit, generally it has some problem in its plugin_init() section.. but P026 has no dedicated plugin_init()

An object property or value may have an unexpected None value. Did you try to delete every json and recreate an empty one?

clumsy-stefan, 2021-03-07:
Up to now it's unpredictable/not reproducable when it happens. It ran now overnight without showing the issue again.

> Is it "P026 'NoneType' object is not subscriptable" ?

I would guess so, yes. It happened when running the "defect" task.

> An object property or value may have an unexpected None value. Did you try to delete every json and recreate an empty one? Editing existing json file may resulted mixed results, especially when RPIEasy running in the background while editing opened files...

No I stopped RPIEasy and edited the json. Basically I just replaced the whole first task with "false" in the json file. Since then it works fine again without showing any issues!

clumsy-stefan, 2021-03-07:
I realised that the sysinfo task (and only that one) is not sending data anymore for some time, even if I use the "run" button in the web interface. Now it looks the following:
[image]

It really seems that something with the sysinfo task is going wrong. I have defined ~30 tasks and all of them seem to run fine, execpt Nr. 1. I try it now with the same definition on a different task.

enesbcs, 2021-03-07:
Please specify your machine, operating system and python version.
As per this screenshot i guess that webform_load() failed to load, maybe a bad taskdevicepluginconfig array value. (none, etc)

You have added several sysinfo devices and only the first one, or only at position 1 is buggy?

clumsy-stefan, 2021-03-07:
As I can't reliably reproduce I can only guess: I added multiple sysinfo plugins but with different tasknames. Up to now I only had issues with task 1 . I moved it now to task 3 (keeping another system info at 2) to see if it happens again.

My system:
Software
--
Build | RPIEasy 2.1.065
Libraries | Python 3.8.8 (default, Feb 20 2021, 21:11:43)[GCC 5.4.0 20160609] Linux-4.4.38-v7+-armv7l-with-glibc2.17
Plugins | 87
Hardware
Type: | RPI-Linux-apt (supported)
OS: | linux 4.4.38-v7+
OS full name: | Ubuntu 20.04.2 LTS
CPU model: | Cortex-A53
CPU speed: | 1200.0000 Mhz
CPU count: | 4
CPU architecture: | armv7l
Hardware: | Pi 3 Model B 1GB
Root access: | True
Python:| 2.7.18

enesbcs, 2021-03-07:
More exception checks added at commit https://github.com/enesbcs/rpieasy/commit/634394e9ce210c0bb449c48f4633366cfc421cf9

enesbcs, 2021-03-07:
> Python:| 2.7.18

Python 2.x is not supported, at any level, although it may work accidentally. 
Please use python 3.x and remove 2.x.

https://www.python.org/doc/sunset-python-2/

clumsy-stefan, 2021-03-07:
sorry... my wrong... It uses python3, so:
```
root@raspi-2:~# python3 --version
Python 3.8.8
```

clumsy-stefan, 2021-03-07:
Hmm.. .after updateing now (essentially stopping, git pull, starting) now my task 2 (first taske used) is not working anymore:

[image]

It won't run and won't edit anymore!

Something gets broken when the service is (unclean) shutdown by the `service` command?

enesbcs, 2021-03-07:
> Something gets broken when the service is (unclean) shutdown by the `service` command?

I am not using the service method, instead the rc.local way, as it enables me to see the full output and restart with CTRL-C with the help of screen command.

> EDIT: I took this assumption as the tasks..json file was written at the time I shutdown the service, even though I hadn't changed anything in the tasks settings for some hours...

Console logs please, from the time of restart and from the time of error.

enesbcs, 2021-03-07:
As i have tested i see that sometimes some tasks.json arrays changes to "null" which causes exceptions in numerous places - but only using the service way. The rc.local method works just fine. As i see, I've made myself a lot of extra work by adding service method...

clumsy-stefan, 2021-03-07:
ok, that would explain, why it does not happen, when I start RPIEasy from the console... So I guess I'll move to rc method..

enesbcs, 2021-03-07:
i've added an extra check on Plugin class init for formula array check, now testing with a P001, P026 and P033, looks like it survived three service restart in a row.

clumsy-stefan, 2021-03-07:
Unfortnately it seems to happen randomly. I also had multiple restarts without it happening and then suddenly it was there again... So I'd rather switch to the obviously "safe" method of rc.local.

enesbcs, 2021-03-07:
Problems can be narrowed by saving tasks.json before restart and after restart. The difference may cause the problem.

clumsy-stefan, 2021-03-07:
I did that the last few times, but then obviously it didn't happen anymore :( as always when trying to pin down an issue... 

But the screen version is obviously nice, as you can connect to the console...

enesbcs, 2021-03-10:
I think the last update solved that, feel free to reopen in case of connected errors.

closed 2021-03-10 by enesbcs

clumsy-stefan, 2021-03-14:
Unfortunately it happened again today, when stopping/starting via service command. Task nr. 2(first task used) was corrupt again.

enesbcs, 2021-03-15:
Json please.