enesbcs/rpieasy #220: [Cosmetics] Clock is only updated on the first task page.

clumsy-stefan, 2021-03-07:
When switching to other task/device pages with `>` the clock at the bottom left is not updated/set anymore (only on the first page it seems to work).

[image]

Nice to have, but not really an issue ;)

enesbcs, 2021-03-07:
javascript magic :D

clumsy-stefan, 2021-03-07:
It also doesn't work on a page full of tasks (if you mean this by "variables"):

[image]

enesbcs, 2021-03-10:
Clock update altered in commit https://github.com/enesbcs/rpieasy/commit/9946ccc9294222c847f5005e2b38a4c66a2bfa64

clumsy-stefan, 2021-03-10:
Working on all pages now for me! Issue can be closed IMHO, thanks a lot!

enesbcs, 2021-03-10:
Closed as solved.

closed 2021-03-10 by enesbcs