enesbcs/rpieasy #221: Has anyone written a driver for SDS011

BNNorman, 2021-03-08:
I like this a lot - it follows closely the Arduino ESPEasy structure - which I have running on several devices.

One omission from the particle devices is the SDS011 which is popular amongst the group I work with (check sensors.connectedhumber.org/app).

I have started by trying to modify the PMS7xxx driver - changing the device to P054 (since the ESPEasy uses that ID) and the name from pms to sds. Next step is to marry up the serial data to the SDS011 then test it.

Are there any tutorials for creating new device drivers that I can read first before I get my first migraine?

Thanks.

tonhuisman, 2021-03-08:
AFAICS the SDS01x sensors are supported by P053, that supports many of those type of sensors.

BNNorman, 2021-03-08:
Really?

The SDS & PMS sensors serial data frames have different formats. Also, the sensor control commands will be different . If you don't mind the sensor being permanently on - although being attached to a Pi that's not a big issue - but longevity of the sensor can be extended by putting it to sleep until needed.  Our sensors transmit data once every 6 minutes to our database so we sleep the sensor till it is time to sample and send data. This was easily done in ESPEasy and the sensors are still working 18 months later.

I installed RPIEasy on a zero but haven't played with it much so maybe I should.

I'll connect an SDS011 to the zero and see how it fares. Will let you know.

Would be handy to know how to create a sensor driver anyway.

Thank you for your response.

BNNorman, 2021-03-08:
Just seen your edit - many thanks.

bsimmo, 2021-03-08:
Yep, that driver needs a better name or some aliases setting up and the 680 taking out. 
Not intuitive , especially if you come from ESPeasy. 

It also does not know the capability of the device and just lists everything any of the set of devices might read. 

There is no on/off/timer feature either.

But it reads PM2. 5 and PM10 when setup OK.

BNNorman, 2021-03-08:
Ben, thanks for your extensive investigation , you beat me to it - perhaps the GPS missing data needs reporting as an issue. Did you have the SDS on as well (two serials?) or maybe the zero can't cope.

I'll have to investigate pypms...

BNNorman, 2021-03-08:
I can see that the limitations of pypms restrict what can be done in RPIEasy.

So, one last suggestion - change the PMSx003 sensor name to something more generic - my first reaction was that it was for plantower only.

closed 2021-03-08 by BNNorman

bsimmo, 2021-03-08:
It on GPIO pins with my setup, I've not tried USB
GPS issue opened.