enesbcs/rpieasy #226: Using service start/stop may result data loss

enesbcs, 2021-03-15:
Hint: Use stable rc.local method until service method proves itself stable.

Budman1758, 2021-03-16:
Not sure if related but today I started up a unit and did a git pull update.  I then went to the device page and everything was gone! Had about 8 devices set up and now the device page is empty. ONLY the devices were gone. All other settings like unit number, name and such were still ok. It was using the service method to start and stop. I never did use that method to stop the service during use. It was only (so far) starting on boot.

Re-installed RPIEasy from scratch. 

Re-added in the first device (BME280) without issues. Trying to add the P082 Position plugin and nothing happens after selecting it. Web interface page just hangs. Open a new tab and the rest of the web interface responds as it should. Going back to the device page that is hung and selected some other plugin and it responds fine. Hit the delete button on that page, went back to main device page and the BME280 device was gone!

One odd thing also. Cannot download the settings json files. no file downloads when clicking the save button. 

Let me know what you need to help...

enesbcs, 2021-03-17:
> Not sure if related but today I started up a unit and did a git pull update. I then went to the device page and everything was gone! Had about 8 devices set up and now the device page is empty. ONLY [...]

Possible, i will test update with service method. 
Now I will mark systemd autostart startup as experimental and not recommended, avoid if possible.

> One odd thing also. Cannot download the settings json files. no file downloads when clicking the save button.

Will try soon.

> EDIT: I can confirm the inability to add the P082 position plugin on 2 separate units. Same behavior on both. Select the plugin and page hangs. Should be separate issue?

Yes, in current RPIEasy bsimmo's P082 version included.
Issue reopened.

Could you try to replace with my last stable version?
https://raw.githubusercontent.com/enesbcs/rpieasy/9946ccc9294222c847f5005e2b38a4c66a2bfa64/_P082_GPS.py

bsimmo, 2021-03-17:
Unless something else happened the only bug with the plug in I could find was present in the old plugins, that is reselecting the plug in option page would stop page updates.
It has debugging in it, so turn that on and see what you can find, all feedback welcome. 

What setup are you using (stick it on the plugins issue page). 
Systemd only? permissions issue, but nothing in that part was changed.. Just the basic code to add debugging and get time from RMC.

enesbcs, 2021-03-17:
Tasks.json saving at shutdown disabled in case of systemd startup type at commit https://github.com/enesbcs/rpieasy/commit/fae69ff8447fd6ee2a2ad3ef2b894d3013e3ef8d

enesbcs, 2021-03-17:
> One odd thing also. Cannot download the settings json files. no file downloads when clicking the save button.

I've tested just now, and it is working on my system.
Make sure that zip is installed and you are running as root.

Budman1758, 2021-03-17:
Seems to work ok after update. Problem seems to be just the initial install of the plugin. Page just hangs after selecting it in the drop-down list.

clumsy-stefan, 2021-03-19:
Switched also back again to systemd, will report if the faulty tasks.json issue reoccurs. thanks for the quick fix!

clumsy-stefan, 2021-03-23:
FI: survived a few start/stops and reboots until now without issues. So currently I would say it's fixed... I'd suggest to reopen if it happens again... thanks a lot for the fix!

enesbcs, 2021-03-23:
Hurray! Thanks for testing!

closed 2021-03-23 by enesbcs