enesbcs/rpieasy #227: GPIO pullup and pulldown not working

Budman1758, 2021-03-21:
I don't know if this is an Rpieasy or a Pi4 problem but it seems that setting pullup or down on gpio pins seems to have no effect. I am trying to get the push button on a rotary encoder working. Trying to set "input-pullup" on the pin but its not pulled up even after reboot. The switch device needs to be "pushbutton active low". There is no response when pushing the switch. Checking voltage on the pin and I'm getting about .5 to .7 volts, Should it not be 3,3 volts set as pullup? Also getting a random voltage when set as pull down.

The config.txt file shows the pins as "input-pu" as it should.  

Also had a problem with another pin not setting as pull down. Wound up using a resistor to pull it to ground. I cannot use this method to pull up with the rotary encoder as the ground for the encoder is shared with the switch.

I have seen some sites about the Pi4 having issues with this but I am under the impression that those issues were resolved. I know that WiringPi needed an update. Is that relevant to this? 

Running out of hair to pull out here.... :>)

enesbcs, 2021-03-21:
Make sure that you are using latest RPi.GPIO 0.7.0
https://pypi.org/project/RPi.GPIO/

You can use pip update on RPIEasy webgui.

Budman1758, 2021-03-21:
> Make sure that you are using latest RPi.GPIO 0.7.0
sudo rpi-eeprom-update
sudo apt update
sudo apt full-upgrade

Latest version installed
Done, done and done.

Everything was already up to date. Got another PI4 8 gigger on the way already. Will see if that changes things.....

enesbcs, 2021-03-22:
Maybe you can try raspi-gpio command line utility on this RPI4. This utility is strictly for debugging purpose, writes directly BCM283x GPIO registers, ignoring anything else that may be using them (like Linux drivers) So it may be wise to stop rpieasy and any application that uses GPIO while testing. 

1/ Install it:

`sudo apt install raspi-gpio`

2/ Set for example BCM21 to Input-Pullup (do not connect to anything yet to BCM21):

`sudo raspi-gpio set 21 ip pu`

3/ Check BCM21 status:

`sudo raspi-gpio get 21`
It will display something like:
GPIO 21: level=1 fsel=0 func=INPUT pull=UP

Which means it is the pullup is enabled and current level is '1', so with a multimeter you has to measure ~3.3V between GPIO21 and GND.
If it is not true, i guess than either the kernel is unable to handle GPIO (i hope you are not using the beta 64bit rpi kernel), or the resistors are gone.

Budman1758, 2021-03-22:
> (i hope you are not using the beta 64bit rpi kernel),

How to tell? I haven't installed a beta version on purpose. 
According to raspi-gpio the pin is pulled up but voltage is not right. Less than 1 volt.

enesbcs, 2021-03-22:
> width: 32 bits
> capabilities: smp
Yes it looks like it is the normal 32bit edition.

> According to raspi-gpio the pin is pulled up but voltage is not right. Only .2 volt. Guess the board is screwed up. Pretty new board and have not really done anything I know of to fry the gpio resistors.....

:( It is very sad. Short circuiting accidentally inputs and outputs without resistors may do bad things. Also placing a small resistor after output pins may increase safety.

Does outputs working normally, just the pullup/pulldown feature is not works?

Budman1758, 2021-03-22:
> Does outputs working normally, just the pullup/pulldown feature is not working?

Actually the pin is not responding at all. I even installed an external pullup and the pin does not respond to a switch input. Seems like a few of the pins on this unit are faulty.

Budman1758, 2021-04-04:
Sorry it took a while to get back on this. Problem seems to have fixed itself with a new board. Guess the old one has gpio issues.

closed 2021-04-04 by Budman1758

enesbcs, 2021-04-05:
I am very sorry about the old board, rest in piece. I hope the new one will work perfectly!

Budman1758, 2021-04-05:
> I am very sorry about the old board, rest in piece. I hope the new one will work perfectly!

The old board is still quite usable as a server or something. Most of the gpio is ok. Just seems to be a few.