enesbcs/rpieasy #228: [Feature] Download link compatibility to ESPEasy

clumsy-stefan, 2021-04-15:
In ESPEasy it's possible to download the stored files (eg. `config.dat`, `rules<n>.txt`) via direct links eg. `http://node.domain/config.dat`.

For backup reasons I did some scripts that go through all nodes and download these config and rules data.

It would be nice if also RPIEasy could be (easily) included in these backups and the files downloaded.

enesbcs, 2021-04-15:
Config files already can be saved at once as a zip file from:  http://IPADDRESS/download?type=settings
But i have to admit, that rules.txt is not yet included in it.

clumsy-stefan, 2021-04-15:
I guess my vote is to have the same URL as for ESPEasy to be able to treat it the same in scripts. Currently I have to differentiate if I'm talking to an RPIEasy node or to a ESPEasy node...

enesbcs, 2021-04-16:
> Currently I have to differentiate if I'm talking to an RPIEasy node or to a ESPEasy node...

Yes, i think differentiate is the best approach as RPIEasy is not ESPEasy, just similar.

From my point of view, there are no 1:1 association between "config.dat" binary file and RPIEasy json text files... 
 
What URL associations do you suggest for advsettings.json, controllers.json, netdev.json, notifiers.json, tasks.json, controllers.json, netman.json and settings.json RPIEasy files to achieve ESPEasy compatibility?

clumsy-stefan, 2021-04-16:
In ESPEasy its always `http://node.name.domain/<filename>` if the filename exists it sends it. I agree that this is not completely collision free, but it works quite smoothly. If the file is not found obviously it doesn't send it and tests if it's actually a command.

enesbcs, 2023-12-26:
added config.dat and rules1.txt download option at commit https://github.com/enesbcs/rpieasy/commit/dc8ccd36c5431463f076c3e2ec65c08e47a79637

closed 2023-12-26 by enesbcs