enesbcs/rpieasy #229: Rule parsing weirdness. Worked before I updated to  2.1.121

Budman1758, 2021-05-12:
I just updated my irrigation controller to the latest version of RPIEasy. It had been some time as I was on version 2.1.0xx something.
Suddenly the rules processor seems to be confused by my setup. This had been working flawlessly for at least a year or more.
Attached is the complete rule file and a copy of the log when the rule is fired.

Basic sequence....  "startwater" gets called.
Task 10 is a dummy device. Value 1 is called "count". This defines which sprinkler valve should be active. This is zero until the "startwater" event sets it to 1. "startwater" then calls the event "autowater"
"autowater" checks the "count" and fires the corresponding "station" event. 
The "station" event is what actually triggers the correct MCPGPIO to turn the water valve on. Each "station" event sets a timer and when the timer ends it updates the "count" by 1 and restarts the "autowater" sequence and this is how the controller moves through each of the water valves.

What's happening now is, when "autowater" is called valve 1 triggers and so do valves 9-16. This lasts about 10 seconds, everything goes off, 10 seconds later that same thing happens again. This results in an endless loop.

Hope you can make sense of my rules file. There is a lot of other stuff in there..... :)

Here are the rules and logs

[log.txt](https://github.com/enesbcs/rpieasy/files/6463063/log.txt)
[rules.txt](https://github.com/enesbcs/rpieasy/files/6463064/rules.txt)

As a side note, is there any easy way to revert to an earlier version?

enesbcs, 2021-05-12:
i will test your code asap.

enesbcs, 2021-05-12:
Event name comparison fixed at commit https://github.com/enesbcs/rpieasy/commit/18d869dc9aa190072d59afb0b403fcdc23095bd9

Budman1758, 2021-05-12:
Well, Looks like you can chalk up another bug squashed. Great job!!

enesbcs, 2021-05-13:
Thanx for testing! This issue popped up after adding the ability to evaluate the same rule names at multiple times.

closed 2021-05-13 by enesbcs