enesbcs/rpieasy #230: P045_MPU6050  alternative address

Epik77, 2021-05-13:
Hello,
i recently installed the i2c rtc_ds1307 module
which takes the address 0x68 used by MPU6050
so I had to change the Gyro sensor address to 0x69.
Although the device is detected:

if I set the address 0x69 to rpieasy, the configuration always forces it to 0x68

One more thing using rpieasy some times I can't write to the rtc_ds1307 module to synchronize the time. Now I use the workaround of stopping the service to synchronize the time, but it would be nice to have it integrated into rpieasy. 

Thanks for the support

enesbcs, 2021-05-13:
Yes, there were a mistype in the P045 code at setting save. Fixed at commit https://github.com/enesbcs/rpieasy/commit/be6bec0b24bc3d37f1bb49fc12f84654ba49c70e

enesbcs, 2021-05-14:
Could you add a bit more detail about how you are using rtc_ds1307?

Epik77, 2021-05-15:
Of course,
the form is the following:
https://it.aliexpress.com/item/32589958767.html?spm=a2g0s.9042311.0.0.27424c4d9tcET9

1pcs I2C RTC DS1307 AT24C32 Real Time Clock Module For AVR PIC ARM Small RTC I2C memory modules DS1307 clock

The module is rtc-ds1307

modprobe rtc-ds1307
inside rc.local:
echo ds1307 0x68> / sys / class / i2c-adapter / i2c-1 / new_device
hwclock --hctosys

the device / dev / rtc0 is created
and you can manage it with the hwclock binary to store or read.

If I start the rpieasy service:

hwclock -r
hwclock: ioctl (RTC_RD_TIME) to / dev / rtc0 to read the time failed: Remote I / O error

I assume the bus is occupied by the I2c LCD display

Having an RTC device gives me the guarantee that the date and time are always synchronized even in the absence of internet.

enesbcs, 2021-05-15:
> modprobe rtc-ds1307
> inside rc.local:
> echo ds1307 0x68> / sys / class / i2c-adapter / i2c-1 / new_device
> hwclock --hctosys
>
> the device / dev / rtc0 is created
> and you can manage it with the hwclock [...]

You are probably right. Although I2C access is not exclusive, but i have no other idea. 
I guess it can be integrated into RPIEasy.
https://github.com/enesbcs/rpieasy/issues/231

> Having an RTC device gives me the guarantee that the date and time are always synchronized even in the absence of internet.

Interesting. I am already using a DS3231 on the same device with RPIEasy, without problems. I guess as i am not writing to it, just leaving the kernel driver to do what it needs with my DS3231.