enesbcs/rpieasy #232: Time comparison in if statement not working

clumsy-stefan, 2021-05-24:
It seems time comparisons in if statements do not work. I have a rule:
```
on Clock#Time=All,**:** do
  if %systime% > 09:00:00
    if %systime% < 19:00:00
      event,UV=on
    else
      event,UV=off
  else
    event,UV=off
  endif
endon
```

The log shows:
```
10:06:00 Event: Clock#Time=Mon,10:06
10:06:00 Parsed condition: if 10:06:00 > 09:00:00 False
10:06:00 Parsed condition: if 10:06:00 < 19:00:00 False
```

Also tried the "simplified" version:
```
on Clock#Time=All,**:** do // switch pool UV-lamp
  if (%systime% > 09:00:00) and (if %systime% < 19:00:00)
      event,UV=on
  else
    event,UV=off
  endif
endon
```

which shows the same, the condition always results in "false". same happens with `%system_hm` and probably others.

While trying I also saw that in ESPEasy the variable is called `%systm_hm` in RPIEasy it's `%system_hm%` which is just a compatibility issue for rules.

enesbcs, 2021-05-25:
>   if %systime% > 09:00:00
> 10:06:00 Parsed condition: if 10:06:00 > 09:00:00 False

You are right, i've never coded time comparison, except the simple event equality testing. Every condition passed directly to python interpreter for evaluation. Its indeed a missing feature.

> While trying I also saw that in ESPEasy the variable is called `%systm_hm` in RPIEasy it's `%system_hm%` which is just a compatibility issue for rules.

Every global variable are in between % and % signs, there are no exceptions - at least in RPIEasy. Regex expressions works fine in that way r"\%([A-Za-z0-9_#\+\-]+)\%", without false positive results, and i have no intentions to break that.

TD-er, 2021-05-25:
> > While trying I also saw that in ESPEasy the variable is called `%systm_hm` in RPIEasy it's `%system_hm%` which is just a compatibility issue for rules.

The variable in ESPEasy is called `%systm_hm%`, so it is not about the `%`, but the extra letter `e`

enesbcs, 2021-05-25:
I see, i can help with the "e". :)

enesbcs, 2021-05-25:
"e" removed in commit https://github.com/enesbcs/rpieasy/commit/7c16098b4e6a044a5364b5f326553eb7207d35a3

clumsy-stefan, 2021-05-25:
Thanks!

The misssing `%`  was a type error from my side, sorry... it was indeed the `e`. Not sure about other system variable inconsistencies, did not compare all of them though.

clumsy-stefan, 2021-05-25:
>You are right, i've never coded time comparison, except the simple event equality testing. Every condition passed directly to python interpreter for evaluation. Its indeed a missing feature.

The same for `%systm_hm%` which obviously also doesn't work... I think it would be a nice and handy feature!

Thanks for looking into this!

enesbcs, 2021-11-23:
Added at commit https://github.com/enesbcs/rpieasy/commit/fadb8031612dd6efbce9efdd9c9f97999a9e8514

clumsy-stefan, 2021-11-25:
Seems to work now!!
```
Nov 25 11:09:50 raspi-2 rpieasy[506]: 11:09:00: Parsed condition: if 11:09:00 >= 11:10:00 and 11:09:00 < 14:00:00 False
Nov 25 11:11:40 raspi-2 rpieasy[506]: 11:10:00: Parsed condition: if 11:10:00 >= 11:10:00 and 11:10:00 < 14:00:00 True
```
Thanks a lot!!! Great work!!

enesbcs, 2021-11-25:
Thanks for testing! Time now converted to seconds from midnight for comparison.

closed 2021-11-25 by enesbcs