enesbcs/rpieasy #233: _P028_BMx280.py reports huge pressure value

vladurash, 2021-05-24:
Installed BME280 which is reporting Temperature and pressure.
On Devices tab only pressure has huge values like 101087.36 or similar. 
Please fix this calculation (I know that I can use formula like %value%/100)

[image]
[image]

Also, from config device page is missing altitude and temperature offset like ESPEasy does for similar plugin.
[image]

Thx

enesbcs, 2021-05-25:
You are right this value seems to high, i will modify the plugin.
Value readings are based upon this library:
https://github.com/cmur2/python-bme280/blob/master/bme280/bme280.py

Honestly there are no functions, variable or any other entry point in this library, that can handle input like "Altitude" so i have to search another unit, if this is needed. My best guess is that this library uses stored calibration data from BME280 for calculations.

About "Temperature offset": this setting is completely redundant, as EasyFormula like %value%+0.2 will work fine, and not just in this plugin, but everywhere.
As RPIEasy is not ESPEasy i have no intentions to make RPIEasy a completely identical clone, its just serving the same/similar purposes.

TD-er, 2021-05-25:
> As RPIEasy is not ESPEasy i have no intentions to make RPIEasy a completely identical clone, its just serving the same/similar purposes.

I'm glad you don't also implement the same bugs as in ESPEasy ;)

enesbcs, 2021-05-25:
> I'm glad you don't also implement the same bugs as in ESPEasy ;)

I am implementing my very own bugs. :)

enesbcs, 2021-05-25:
pressure divided by 100 at commit https://github.com/enesbcs/rpieasy/commit/7c16098b4e6a044a5364b5f326553eb7207d35a3

vladurash, 2021-05-25:
I agree that RPIEasy is not ESPEasy. Maybe a tips and tricks how to translate plugins from arduino to python would be more attractive for new joiners on this nice project.

closed 2021-05-25 by vladurash