enesbcs/rpieasy #234: eventvalue re-written

Amonbofils, 2021-06-01:
Hi,

i found something wrong. In a rule i wrote that script :

// Request events for fermenting tank 2
On setTemp_2 do
    TaskValueSet 5,2,%eventvalue%
    TaskValueSet 14,2,%eventvalue%+0.5
    TaskValueSet 15,2,%eventvalue%-0.1
endon

What i wanted, and i suppose it has to work is : 

If eventvalue is like 20, then TaskValueSet 14,2 is supposed to be 20.5 and taskValueSet 15,2 is supposed to be 19.9.

But, really it's 14,2 at 20.5 (that's ok) and 15,2 is 20.4. 

eventvalue seems to be re-written by the TaskValueSet insctuction.

Is it normal or wrong ?

Thanks

enesbcs, 2021-06-15:
This behaviour is a result of the requested feature to add taskvalues as eventvalues.. which i think will restore to its original state as it is not good.

TD-er, 2021-06-15:
> This behaviour is a result of the requested feature to add taskvalues as eventvalues.. which i think will restore to its original state as it is not good.

You should keep the eventvalues and the events together.
In ESPEasy I create a string for an event, including the eventvalues. So no matter if it is put in a queue, or dealt with immediately, the events and eventvalues stay together and don't change.

enesbcs, 2021-11-23:
Reverted back to original RPIEasy behavior, at commit https://github.com/enesbcs/rpieasy/commit/fadb8031612dd6efbce9efdd9c9f97999a9e8514 taskvaluset do not overwrite eventvalues anymore.
Closing.

enesbcs, 2021-12-21:
https://github.com/enesbcs/rpieasy/issues/219#issuecomment-998287172

closed 2023-12-25 by enesbcs