enesbcs/rpieasy #235: Ili9341

cybermandiant, 2021-06-04:
Hi I will want to test with a tft ili9341 2.4 i have activ spi in my rpi and on rpieasy I can not find how to connect it someone can tell me what the enable pin is because I do not have one on my screen I have dc miso mosi sck cs led vcc gnd reset Thank you

tonhuisman, 2021-06-04:
`Enable` -> `cs` (read: 'chip select')

For a clear view you will have to connect `led` to `vcc`, I expect, as led is usually the backlight.

cybermandiant, 2021-06-04:
Thanks for your fast reply i have try this... Not work i have connect bl directly to 3.3v

cybermandiant, 2021-06-04:
Yes led is my backlight

cybermandiant, 2021-06-04:
Is work perfectly in esp32 not work with my rpi zero w

cybermandiant, 2021-06-04:
Yes is working i have reset all my raspberry

cybermandiant, 2021-06-04:
Thanks a lot for all :)

closed 2021-06-04 by cybermandiant