enesbcs/rpieasy #237: P2P Network didnt work

crazybanane, 2021-06-18:
Hello i got a little problem,

I have a RPI Zero W and some ESPs i installed now DietPi on the Zero and configured everything that it works as Hotspot.

I can connect the ESP to RPI with a stable connection and when i configure P2P networking i see on RPI the ESP Node but i didnt see the RPI node on ESP and when i configure and activate on ESP an Device on Tasklist and i want to send it to RPI then the Task wont show in the list on RPI. What can i do or what do i wrong? Every Wifi connection works over DHCP.

Pls help

tonhuisman, 2021-06-19:
Do all units/nodes have a unique id between 1 and 254?

crazybanane, 2021-06-19:
Yes all units got unique id. On RPi in Log there is shown:
00:11:13 New P2P unit discovered: 3 192.168.4.84 50:02:91:79:A2:63

but on RPI under Devices no input is shown.

UDP is actually on both sides on 8266 port

enesbcs, 2021-06-19:
> Yes all units got unique id. On RPi in Log there is shown:
> 00:11:13 New P2P unit discovered: 3 192.168.4.84 50:02:91:79:A2:63
> 
> but on RPI under Devices no input is shown.

So P2P network is working, but Devices which marked as shared on the ESP side not seen on the RPI side?

crazybanane, 2021-06-19:
If i try P2P via a normal Wifi adapter/router then they can see each other on both sides there are listed 2 nodes as seen here.
https://ibb.co/3ptj3Cb
so the Nodes see each other works fine
https://ibb.co/dJdBMqY

So if i Activate AP on RPI and connect ESP to it then it looks like that only RPI can see ESP and in log there is P"P discovery Schown for ESP unit 3 but not as itsself.
and the connection disappears tried with reboots or reset controllers and make new.....

see here:
https://ibb.co/zGSkC51
and:
https://ibb.co/qYbzH4H

enesbcs, 2021-06-20:
> and the connection disappears tried with reboots or reset controllers and make new.....
> 
> see here:
> https://ibb.co/zGSkC51

It looks like a basic network addressing question.

According this picture the RPI AP IP address is 192.168.4.1, but the shown IP address of the primary network is 192.168.43.107 on the status page. Try to change the primary interface at config page or disable the other interface, as P2P networking works only on the primary network. (well honestly it does an UDP broadcast to everybody, but if its advertising itself as .43.107 IP on that, than it will surely not getting an answer from anybody on .4.x network)

As an enhancement a select box may be implemented on the P2P controller page, to let the user select the necessary network interface.

crazybanane, 2021-06-20:
Ok i will try to edit config, but i have only Primary Wifi as first instance of installation for software and the system after that i disable and disconnect and let the Wifi-Hotspot/AP-Mode running. But when i try to zero or leave the config page blank everytime the system reboot. then there are the same IPs shown that are seen, even if i try to clean /etc/network/interfaces file same result the IPs will be automaticly recovered.

So do you have a workaround?

Thanks for your great help.

enesbcs, 2021-06-20:
try to delete data/netman.json

crazybanane, 2021-06-24:
Hey guys sorry this week was strong so i tried now several things but no work.
Anytime i can start as AP, RPIEasy shows on start site.

WLAN-RSSI: Wired Connection
 IP-Adress: 192.168.4.1

Then it doenst shows hisself as a Node.

I deleted the files you said but nothing wont work.

crazybanane, 2021-06-27:
Hello this is the list what i made step by step:
Written line by Line

HW: Raspberry PI Zero W

Software DietPI 7.3.2

Flashed via Balena Etcher

Setting up dietpi-wifi.txt and dietpi.txt

>insert SD Card boot via HDMI and Keyboard......

check for correct installation

then via Terminal:

>sudo apt install python3-pip screen alsa-utils wireless-tools wpasupplicant zip unzip git

>git clone https://github.com/enesbcs/rpieasy.git

>cd rpieasy

>sudo pip3 install jsonpickle

>reboot

>cd rpieasy

>sudo ./RPIEasy.py

LOG:

Starting RPIEasy 2.1.166
Init hardware...
Load network settings...
/bin/sh: 1: ifconfig: not found
Load GPIO settings...
RPi.GPIO not installes!
I2C smbus not installed!
GPIO init failed
Creating new pinout
Load devices from file
Load controllers from file
Load notifiers from file
/bin/sh: 1: ifconfig: not found
07:41:46: Webserver starting at http://False:80

LOGEND

>reboot

>sudo apt install net-tools

>cd /rpieasy

>sudo ./RPIEasy.py

LOG:

Starting RPIEasy 2.1.166
Init hardware...
Load network settings...
Load GPIO settings...
RPi.GPIO not installed!
I2C smbus not installed!
GPIO init failed
Creating new pinout
Load devices from file
Load controllers from file
Load notifiers from file
07:50:24: Webserver starting at http://192.168.43.107:80

LOGPAUSE

Connect via PC to >RPi
TAB Config
Set Unit Number to 1
Install wifiap

LOGON

07:57:06: Settings saved without OS network settings modifications as you wish!
07:57:48: apt: apt-get update && apt-get install -y hostapd dnsmasq
debconf: delaying package configuration, since apt-utils is not installed
08:01:07: !`pip`
08.01:07: exec: systemctl disable dnsmasq hostapd && systemctl unmask hostapd
Synchronizing state of dnsmasq.service with SysV service script with /lib/systemd/systemd-sys-install.
Executing: /lib/systemd/systemd-sysv-install disable dnsmasq
Synchronizing state of hostapd.service with SysV service script with /lib/systemd/systemd-sys-install.
Executing: /lib/systemd/systemd-sysv-install disable hostapd
Unit /etc/systemd/system/hostapd.service is masked, ignoring.
Removed /etc/systemd/system/multi-user.target.wants/dnsmasq.service
Removed /etc/systemd/system/hostapd.service

LOGPAUSE

RPIEAsy
under Plugins install GPIO

LOGON

08:09:32: apt: apt-get install update && apt-get install -y python3-dev python3-pip python3-setuptools
debconf: delaying package configuration, since apt-utils is not installed
08:13.46: pip3: sudo -H pip3 install RPI.GPIO
08:14:22: !'installcmd'
Failed to get unit file state for rpieasy.service: No such file or directory

LOGPAUSE

RPIEasy
under config

Start AP when: AP startup without condition
On this device: First wireless network device
After this time: 5 sec
Stop Ap after: Never

Submit

LOGON:
08:24:56: Settings saved without OS network settings modifications as you wish!
LOGPAUSE

Deactivate Router-Wifi

>reboot

>cd rpieasy
>sudo ./RPIEasy.py

LOGON:

Starting RPIEasy 2.1.166
Init hardware...
Load network settings...
Stopping network services...
Failed to stop dhcp.service: Unit dhcp.service nt loaded.
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface..
Starting DNSMASQ server...
RTNETLINK answeres: File exists
Starting AP on wlan0...
Load GPIO settings...
I2C smbus not installed!
Creating new pinout
Load devices from file
Load crontollers from file
Load notifiers from file
08:28:31: Webserver starting at http://192.168.0.100.80

LOGPAUSE

PC connect to Webserver

cant connect via 192.168.0.100:80 must connect via 192.168.4.1 

Unit:	1
Local Time:	2021-06-27 08:39:36
Uptime:	0 days 0 hours 11 minutes
Load:	3.45 %
Free Mem:	450752 kB
IP:	192.168.0.100
Wifi RSSI:	Wired connection
Build	RPIEasy 2.1.166

try a reboot

same thing

on Webserver under Controller
set 1 to ESPEasy P2P

Port 8266

reboot

in Webserver under Hardware set RPIEasy autostart at boot with systemctl (experimental, not recommended): X
because the other option did not work.

reboot

I found no way to get Node/P2P networking available 
if you need more informations or other LOG files then call
Hope this will help

[image]
[image]

crazybanane, 2021-06-27:
So after long time of testing and many fails i found a workaround to get it work as AP without Internet connection but with P2P connection and all other functions that didnt need Internet.

I write now only for RPI Zero W

Download official raspbian on [here](https://www.raspberrypi.org/software/operating-systems/)

Flash via BalenaEtcher.
Insert Sd card in RPI and connect to HDMI.
After boot set RPI up for normal wifi connection and do your settings in raspi-config.
Ready?..... 
Now you NEED Raspap!
In terminal type:
`curls -sL https://install.raspap.com | bash`

The first 2 Questions answer with Y
If it is asking for install adblock say No
If it is asking for install OpenVPN say No

When ready then follow the normal instructions but with Net-tools!!

`sudo apt install python3-pip screen alsa-utils wireless-tools wpasupplicant zip unzip git net-tools
git clone https://github.com/enesbcs/rpieasy.git
cd rpieasy
sudo pip3 install jsonpickle`

now STOP

When this is all installed deactivate the normal Wifi from your router or connection that you used for setup and reboot RPI in terminal!

Now wait until RPI is booted up and you see an Wifi with name: raspi-webgui on your smartphone or notebook etc and connect to them.

Try to reach the Webserver normaly under 10.3.141.1

If you see Webserver from RaspAP go to System>Advanced and change Webserver Port from 80 to 8080 and save.

Now reboot the last time.

In Terminal you can do now the Setup for RPIEasy

`cd /rpieasy
sudo ./RPIEasy.py`

then you can follow the normal steps.

RPIE > Set unit Number to 1> Controllers set P2P networking and change Port to 8266> Connect any other ESPEasy or RPIEasy to the raspi-webgui Wifi and do the Steps for the P2P connection. Dont forget to set in ESPEasy under Settings>Advanced under UDP Port 8266!!

enesbcs, 2021-11-23:
IP address now can be fixed manually for example to 192.168.4.1 at P2P plugin page from commit https://github.com/enesbcs/rpieasy/commit/fadb8031612dd6efbce9efdd9c9f97999a9e8514

closed 2022-05-29 by enesbcs