enesbcs/rpieasy #238: Neopixel pwm problem ?

klem1, 2021-07-04:
Hi sorry for my english i am french.

I have a problem, I'm trying to connect a 10 led neopixel strip on my raspberry pi 3. I have connected the 5v strip leds to an external power supply and the DATA pin to the gpio 18.
&

When I test a command like this "NeoPixelAll, 255,0,0,255" my strip led does not change to the right color directly, I have to press the submit key several times for it to take the correct one into account color. I have the impression that the pwm is not working.

[image]
[image]

I forgot something ?

i will upload the video in 1 hour

enesbcs, 2021-07-04:
when you are using 5V supply for an external device, i hope you are using a Level shifter between the GPIO and the 5V device DATA pin, unless you want to destroy your RPI.

klem1, 2021-07-04:
i dont use level shifter i follow this :/
https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-WS2812-Steckplatine-600x361.png.

klem1, 2021-07-07:
i put a diode like this, but its the same.
[https://learn.adafruit.com/assets/64122](url)

klem1, 2021-07-15:
u can close it, i found the problem i put a 74AHCT125 level converter chip, and it's perfect now.

closed 2021-07-15 by klem1