enesbcs/rpieasy #239: dasboard.esp does not do conversions

clumsy-stefan, 2021-07-18:
Not really a big issue, but `dashboard.esp` is only replacing values of the form `[TaskName#ValueName]` and not the enhanced version `[TaskName#ValueName#transformation#justification]` as ESPEasy does for easy calculations. So `dashboard.esp` needs to be adpated accordingly when transferred from an ESPEasy to RPIEasy device.

Also `%syslaod% ` seems not to be known as system variable.

enesbcs, 2021-11-25:
Are there any manual available about this enhanced version format?

clumsy-stefan, 2021-11-25:
Here:
https://espeasy.readthedocs.io/en/latest/Rules/Rules.html
Look for "Formatting refered values" (about 1/3rd down the page)

TD-er, 2021-11-25:
> Also `%syslaod% ` seems not to be known as system variable.

What about `%sysload%` ? ;) (not sure if it exists in RpiEasy, just noticed the swapped letters)

enesbcs, 2021-11-25:
> What about `%sysload%` ? ;) (not sure if it exists in RpiEasy, just noticed the swapped letters)

System load exists in P026 Sysinfo plugin in RpiEasy. It can be added as a global variable, but calling this too soon (every second) will result very heavy cpu loads, this is the reason why i omit it.

enesbcs, 2023-12-25:
Please show me some real world examples, i still have no clue what are the 'justifications'