enesbcs/rpieasy #240: Raspberry pi 1 B model

BIIINO, 2021-08-01:
Hello,

i am using rpi 1 model B and everywhere  (command line and webui) and i have this issue:

!Your GPIO can not be identified!

How can i fix it please ? thank you very much in advance

enesbcs, 2021-08-02:
Please show me the result of this command, and i will test it on my installation with your HW ID:

cat /proc/cpuinfo

BIIINO, 2021-08-02:
Hello, se bellow, thank you

`root@raspberrypi:/home/pi# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 100000e
Serial          : 0000000061580c8d
Model           : Raspberry Pi Model B Rev 2
`

BIIINO, 2021-08-04:
I try to replace in:
/rpieasy/**linux_os.pylinux_os.py**

(hwid == "1000e")   with (hwid == "100000e")  but when i start it again, nothing happen, still same issue.

enesbcs, 2021-08-04:
it may help if you stop RPIEasy and delete json files in data directory, than restart RPIEasy to detect new board.

BIIINO, 2021-08-04:
I remove whole rpieasy folder, reboot pi, clone it again from github, edit linux_os.pylinux_os.py with current hwid, start and nothignh changes... Where is that folder what you mean ? in /rpieasy/data i have only readme.txt and nothing else.

Thank you

BIIINO, 2021-08-05:
Formatted rpi sdcard, fresh install raspbian, replaced (hwid == "000e") with (hwid == "100000e")  and still issue. What else can i try to fix it ?

BIIINO, 2021-08-06:
problem is with:

     if hwid[:4] == "1000":
      hwid = hwid[:-4]

in linux_os.py , when i remove these two lines and manualy add  ----or  (hwid == "100000e")--- rpi is recognized and everything work as excepted... i try another combination with this calculation but without success

enesbcs, 2021-08-06:
fixed in commit https://github.com/enesbcs/rpieasy/commit/6546f728a47c988d7387b59374ab79931183703a

BIIINO, 2021-08-07:
RPI 1 B rev2 detected - looks good

closed 2021-08-20 by enesbcs