enesbcs/rpieasy #241: rpi0w with envirophat i2c

docsparks, 2021-08-04:
https://shop.pimoroni.com/products/enviro-phat

device is a stack on hat for rpi0w,  claims to use i2c, and sensors recognized by rpieasy,  does any one have any advice how id go about implementing into rpieasy? thanks

enesbcs, 2021-08-04:
bmp280 and ads1015 already supported
tcs3472 and lsm303d support can be added using official python example from here:
https://github.com/pimoroni/enviro-phat/tree/master/library/envirophat

docsparks, 2021-08-04:
i do appreciate your response, but even when i select bmp280 and choose i2c-1 (to use gpio 2/3) im not getting anything within the rpieasy setup.  not sure what i could be doing wrong. long time user first time caller, thanks again

docsparks, 2021-08-04:
ohh shit, after updating all packages and changing address, it works,  i stand corrected and have more fucking around to do, thank you.

closed 2021-08-05 by docsparks