enesbcs/rpieasy #242: Switch input does not create an event?

Budman1758, 2021-08-20:
I'm a bit confused here. 

I have a simple PIR sensor switch connected to a GPIO pin. The pin is configured as an input in the pin & ports page. On the Devices page you can see the state change when the sensor is triggered. However there is no event triggered in the log and the command I'm attempting to run does not run.

In the device setup page the switch is set to detect both rising and falling, and no long press action. But no event generated at all.
If I set the pin as an output I get this message in the log.  "Event can not be added, register backup timer You must setup() the GPIO channel as an input first"  and it stops reacting at all. No state change in the devices page.

What am I doing wrong here?

enesbcs, 2021-08-20:
If the error log is saying that it is not an input, than i guess it is not an input. Maybe something else is uses or reconfigures that pin. 
When you check Hardware/pinout & ports page on the specific row you can see "Detected function" column which shows the actual state of the pin. In case of problems, please send the json settings file to debug.

enesbcs, 2021-08-20:
Please send the complete data directory for debugging.

Budman1758, 2021-08-20:
[data.zip](https://github.com/enesbcs/rpieasy/files/7024253/data.zip)

Here is the data file. The pin is configured and detected as an input. Pin in question is GPIO 19

enesbcs, 2021-08-21:
Sorry i am unable to reproduce this issue.
I have used the json files you provided, and added the following into Rules:

```
on pir#switch do
 if [pir#switch]=1
  taskvalueset 5,1,1
 else
 taskvalueset 5,1,0
 endif
endon
```
The event fired on both 0 and 1 directions according to log:

```
08:18:36: Event: pir#switch=0
08:18:36: CMD: taskvalueset 5,1,0
08:18:36: OS command started at background
08:18:36: Event: pir#switch=1
08:18:36: CMD: taskvalueset 5,1,1
08:18:36: OS command started at background
```

I've changed "input" to "input pullup" on GPIO19 and used a simple push button for testing. Although it has no relevancy.

I see that GPIO5 and 6 are output pins, i prefer not to use Input switch type device, instead P029 Output helper for output pins.

Budman1758, 2021-08-21:
Now I feel pretty silly. I did not have a rule for the switch. I was trying to use the trigger variable function in the Run OS Command device to send a command to the camera software on the PI. 

So unless there is a rule for the input-switch function there is no event generated?

enesbcs, 2021-08-21:
Indeed if the Rules is empty, there will be no event generated at all.

But P511 works with other way, it does not lookup events, it checks directly the selected task value changes 10 times per second. So i have to check P511. Altering issue name to avoid confusions.

enesbcs, 2021-08-22:
@Budman1758 in your configuration the trigger of the P511 plugin is the task named "irarray" which is an output, and i have no idea what it does.
The input named "PIR" is Task4 and when i've added this to the P511 plugin as a trigger, it works.

```
10:29:18: OS command started at background
/bin/sh: 1: cannot create /var/www/FIFO: Directory nonexistent
10:29:18: OS command started at background
XXX  echo 'im' >/var/www/FIFO  XXX
```

Although the apostrophe is not escaped in the command, so it is lost upon saving. Fixed by urlencoding commands in commit https://github.com/enesbcs/rpieasy/commit/04023e5d3155c8a9a65000d9b1f56cbf78ca4bb7

Budman1758, 2021-08-22:
At first I had the "irarray" task as trigger but I changed it right after I sent that data file. That's for a IR LED light module. 

The trigger is now working as it should without a rule and the command with the apostrophe works and saves as it should.  Thanks.

Looks like this issue can be closed. Great work as usual!

enesbcs, 2021-08-22:
Thank you for testing!

closed 2021-08-22 by enesbcs