enesbcs/rpieasy #243: OpenTherm support?

vampywiz17, 2021-08-20:
Hello there,

It possible to add OpenTherm support, like ESPeasy do this?

https://github.com/okar1/ESPEasy-OpenTherm

https://github.com/ihormelnyk/opentherm_library

enesbcs, 2021-08-21:
I guess it is possible to add, if anyone has an opentherm device and a lots of free time.
Unfortunately I do not have any of the above.

vampywiz17, 2021-08-21:
I have this device and I able to help you.

enesbcs, 2021-08-21:
a device which clones me would help a lot :D

vampywiz17, 2021-08-21:
> a device which clones me would help a lot :D

I understand you :)

TD-er, 2021-08-21:
> a device which clones me would help a lot :D

If you have found one, I want one too!

Budman1758, 2021-08-21:
> a device which clones me would help a lot :D

Count me in on that. Perhaps a younger version would be nice also.....

Yocee84, 2022-01-01:
This not help?
https://github.com/okar1/ESPEasy-OpenTherm

enesbcs, 2023-12-25:
Closed as stalled

closed 2023-12-25 by enesbcs