enesbcs/rpieasy #248: Update to RPIEasy 2.1.327  rules not working

AtyKv, 2021-12-01:
Hi ,
After update my rules not working . The test email its ok..  

some example 

On System#Boot=1 do 
notify 1, RPI Easy Boot
endon
 
On Door#switch=1 do
notify 1, Opendoor
endon

On Bell#switch=1 do
notify 1, BingBang
endon

enesbcs, 2021-12-02:
Can you try previous commands.py?
https://raw.githubusercontent.com/enesbcs/rpieasy/7c16098b4e6a044a5364b5f326553eb7207d35a3/commands.py

AtyKv, 2021-12-02:
How to do that  ?   Some help in Rpieasy  pls .

AtyKv, 2021-12-02:
How to install the previous commands.py ? thx :)

AtyKv, 2021-12-03:
Menege to replace the commands.py file . The rules stil not working .

AtyKv, 2021-12-03:
found this in logs
10:56:08 Event can not be added, register backup timer You must setup() the GPIO channel as an input first
10:56:08 Event can not be added, register backup timer You must setup() the GPIO channel as an input first

enesbcs, 2021-12-03:
Then it is not rules related problem, i guess you are using two P001 Switch input plugins. From which RPIEasy version do you upgraded, as P001 did not updated at latest commit?

AtyKv, 2021-12-03:
I dont now what version i have . This pi is installed in my electrical board  for almoust a year.  Works fine until 14 nov .when i mayde the 3 updates from update menu . Now only the test message arrive to my email.  The simple On System#Boot=1 do notify 1, RPI Easy Boot endon  command not [work.](url
[image]
)
[image]

enesbcs, 2021-12-04:
> 10:56:08 Event can not be added, register backup timer You must setup() the GPIO channel as an input first
> 10:56:08 Event can not be added, register backup timer You must setup() the GPIO channel as an input first

According the screenshots, this came from Relay1 and Relay2 as they are not input types. Output types has a dedicated Output helper plugin FYI. But this is not related to the problem.

Did you see any change in Door or Bell Values on the Device page when pin status changes? If not, then it is not Rule related, instead P001 related. P001 has changed a few times in the last year, please enter plugin settings using "Edit" button, make sure, that Long press disabled, use "BOTH" as trigger mode, press submit and retest.

AtyKv, 2021-12-04:
Yes it change .I can command the switches from distance . The status is ok
.  Only the notifications not working .All other functions and timers are
ok .

AtyKv, 2021-12-05:
On System#Boot=1 do
notify 1, RPI Easy Boot
endon

This simple command is not working at all .  In command window if i type
notify 1 the email is working fine .

tonhuisman, 2021-12-05:
System#boot may be to early to send anything, as WiFi is not fully functional yet.

`on WiFi#Connected do`  would probably a beter event to send out anything via... WiFi.

AtyKv, 2021-12-05:
On System#Boot=1 do 
notify 1, RPI Easy Boot
endon

On WiFi#Connected=1 do
notify 1 , wifi connected 
endon
 
On Door#switch=1 do
notify 1, Opendoor
endon

Try de wifi command .....   not working   no notification

tonhuisman, 2021-12-05:
I'm not sure if the WiFi#Connected uses a 1 as the first argument, so you might need to leave the `=1` part out.

AtyKv, 2021-12-05:
On System#Boot=1 do 
notify 1, RPI Easy Boot
endon

On WiFi#Connected do
notify 1 , wifi connected 
endon

Nope  .. :(

tonhuisman, 2021-12-05:
> so you might need to leave the `=1` part out.

And the same is probably true for the `System#Boot` event. At least on ESPEasy it doesn't have any arguments.

AtyKv, 2021-12-05:
Before the update of the rpieasy (14nov) all the  rules  works fine .  I receve all the notifications .

AtyKv, 2021-12-05:
If the problem cant be solved is any way to downgrade to a previous version ?

tonhuisman, 2021-12-05:
Not sure about the script parsing in RPIEasy, but ESPEasy is somewhat strict on use of commas and spaces as separators, so you might need a little cleanup, and I also added quotes around the arguments that have spaces:
```
On System#Boot do
notify 1,"RPI Easy Boot"
endon

On WiFi#Connected do
notify 1,"wifi connected"
endon
```

NB: I haven't tested this on my RPIEasy, that one ended up in semi-longterm storage, so it takes some time to get that running.

> Before the update of the rpieasy (14nov) all the rules works fine

Hm, that's quite easy to investigate for @enesbcs I guess, (my Python skills are very limited)

I don't know how to downgrade the RPIEasy installation, sorry.

enesbcs, 2021-12-05:
> If the problem cant be solved is any way to downgrade to a previous version ?

Yes, github has all versions. But you have to know which version do you want to use...

enesbcs, 2021-12-05:
> The wifi command ..... not working no notification

Yes, as wifi managed by the Linux Operating System and it 99% of the cases it was on, before RPIEasy starts up...

enesbcs, 2021-12-05:
Did you use inverted type on P001 Switch input plugin? As i see the GPIO type is input-pullup, but its value is 0..
If possible please send me your json file for debugging.

tonhuisman, 2021-12-05:
> as wifi managed by the Linux Operating System and it 99% of the cases it was on, before RPIEasy starts up...

In that case I'd have expected at least 1 time trigger of that event, if only for some backward-compatibility with ESPEasy, that will send that event once WiFi is connected. A lot of rules-scripts may expect that.

From reading the last couple of commits I've not detected anything that could break this, but some parts are somewhat hard to understand.

AtyKv, 2021-12-05:
Can you help me with the command line syntax ? sudo apt get install ??**     from github to install some version befor 13nov . I don't now how to identify .

enesbcs, 2021-12-05:
github has a zip archive for every commit... easiest to download and uncompress it manually https://github.com/enesbcs/rpieasy/archive/04023e5d3155c8a9a65000d9b1f56cbf78ca4bb7.zip

enesbcs, 2021-12-05:
For first sight "On System#Boot=1 do" has no sense, "On System#Boot do" on the other hand works. I have to fix that.

enesbcs, 2021-12-05:
It may solved by commit https://github.com/enesbcs/rpieasy/commit/6d1af9b40f8f0d956dc4bbd150ac9f86a381d5ac

AtyKv, 2021-12-05:
Yesss     Thank you all :)  . Replacing this two files solve the problem ..   Have notifications again .

enesbcs, 2021-12-05:
> > as wifi managed by the Linux Operating System and it 99% of the cases it was on, before RPIEasy starts up...
> 
> In that case I'd have expected at least 1 time trigger of that event, if only for some backward-compatibility with ESPEasy, that will send that event once WiFi is connected. A lot of (ESPEasy-converted) rules-scripts may expect that.

But its not ESPEasy. And there are even no guarantee, that a wifi device exists under RPIEasy, as it also works with wired LAN alone on an RPI3 or on a pure x86 Linux PC with wires only. :)

> 
> From reading the last couple of commits I've not detected anything that could break this, but some parts are somewhat hard to understand.

I guess that this bug is related with a fix at 2021 March which deals with similar Event names. (event, event1, event111...) I there were no equal sign or at least there were on space before equal sign, this error wont popped out.

closed 2021-12-05 by enesbcs