enesbcs/rpieasy #250: Cannot add a shelly device

hansa031, 2022-04-02:
I have rasbian buster with domoticz 2022.1 running on my rpi 4 and when i installed shelly_mqtt plugin in domoticz i get the following error:
 2022-04-02 22:04:06.235 Status: Shelly: Error adding device: shellyswitch25-3494546A09CD-input_event0
2022-04-02 22:04:06.235 Status: Shelly: Error adding device: shellyswitch25-3494546A09CD-input_event1
2022-04-02 22:04:06.838 Status: Shelly: Error adding device: shellyswitch25-98CDAC250170-input_event0
2022-04-02 22:04:06.839 Status: Shelly: Error adding device: shellyswitch25-98CDAC250170-input_event1 
As result of this i don't see the shelly's in the device list.

But the logging of the plugin is working:
 2022-04-02 22:26:21.062 Shelly: Heartbeating...
2022-04-02 22:26:21.062 Shelly: MqttClient::ping
2022-04-02 22:26:22.416 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/0 off
2022-04-02 22:26:22.416 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-0
2022-04-02 22:26:22.416 Shelly: -1
2022-04-02 22:26:22.416 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/0/power 0.0
2022-04-02 22:26:22.416 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-0-energy
2022-04-02 22:26:22.417 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/0/energy 0
2022-04-02 22:26:22.417 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-0-energy
2022-04-02 22:26:22.417 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/1 off
2022-04-02 22:26:22.417 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-1
2022-04-02 22:26:22.417 Shelly: -1
2022-04-02 22:26:22.417 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/1/power 0.0
2022-04-02 22:26:22.417 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-1-energy
2022-04-02 22:26:22.417 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/relay/1/energy 0
2022-04-02 22:26:22.417 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-1-energy
2022-04-02 22:26:22.418 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/input/0 0
2022-04-02 22:26:22.418 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-0-input
2022-04-02 22:26:22.418 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/input/1 0
2022-04-02 22:26:22.418 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-1-input
2022-04-02 22:26:22.418 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/temperature 36.56
2022-04-02 22:26:22.418 Shelly: >>> Looking for device: shellyswitch25-3494546A09CD-temp
2022-04-02 22:26:22.418 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/temperature_f 97.8
2022-04-02 22:26:22.419 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/overtemperature 0
2022-04-02 22:26:22.419 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/temperature_status Normal
2022-04-02 22:26:22.419 Shelly: MQTT message: shellies/shellyswitch25-3494546A09CD/voltage 228.64
2022-04-02 22:26:23.171 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/0 off
2022-04-02 22:26:23.171 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-0
2022-04-02 22:26:23.171 Shelly: -1
2022-04-02 22:26:23.172 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/0/power 0.0
2022-04-02 22:26:23.172 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-0-energy
2022-04-02 22:26:23.172 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/0/energy 0
2022-04-02 22:26:23.172 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-0-energy
2022-04-02 22:26:23.172 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/1 on
2022-04-02 22:26:23.172 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-1
2022-04-02 22:26:23.172 Shelly: -1
2022-04-02 22:26:23.172 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/1/power 0.0
2022-04-02 22:26:23.172 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-1-energy
2022-04-02 22:26:23.173 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/relay/1/energy 0
2022-04-02 22:26:23.173 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-1-energy
2022-04-02 22:26:23.173 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/input/0 0
2022-04-02 22:26:23.173 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-0-input
2022-04-02 22:26:23.173 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/input/1 1
2022-04-02 22:26:23.173 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-1-input
2022-04-02 22:26:23.173 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/temperature 43.01
2022-04-02 22:26:23.173 Shelly: >>> Looking for device: shellyswitch25-98CDAC250170-temp
2022-04-02 22:26:23.173 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/temperature_f 109.42
2022-04-02 22:26:23.174 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/overtemperature 0
2022-04-02 22:26:23.174 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/temperature_status Normal
2022-04-02 22:26:23.174 Shelly: MQTT message: shellies/shellyswitch25-98CDAC250170/voltage 230.04
2022-04-02 22:26:30.725 P1 Wifi gateway: P1 Smart Meter (Electra)
2022-04-02 22:26:30.735 P1 Wifi gateway: General/Voltage (Voltage L1)
2022-04-02 22:26:30.744 P1 Wifi gateway: General/Voltage (Voltage L2)
2022-04-02 22:26:30.753 P1 Wifi gateway: General/Voltage (Voltage L3)
2022-04-02 22:26:30.761 P1 Wifi gateway: Usage (Usage L1)
2022-04-02 22:26:30.770 P1 Wifi gateway: Usage (Usage L2)
2022-04-02 22:26:30.778 P1 Wifi gateway: Usage (Usage L3)
2022-04-02 22:26:30.787 P1 Wifi gateway: Usage (Delivery L1)
2022-04-02 22:26:30.792 P1 Wifi gateway: Usage (Delivery L2)
2022-04-02 22:26:30.796 P1 Wifi gateway: Usage (Delivery L3)
2022-04-02 22:26:31.088 Shelly: Heartbeating...

Can you please help me? THX!

hansa031, 2022-04-02:
Sorry i cannot comment on shelly_mqtt repos. thats why try it this way sorry ;)

enesbcs, 2022-04-03:
It is no coincidence that Issues has been disabled on shellymqtt page. You can try to reach the forum for community support.
https://www.domoticz.com/forum/viewtopic.php?p=288383#p288383

The error message you send is from plugin.py line 823 and it is because it failed to create a new device, either because you disabled automatic device creation in Domoticz main settings, or reached the maximal 254 device number, or the Domoticz Python API changed a lot...

closed 2022-04-03 by enesbcs