enesbcs/rpieasy #252: Not enough Devices for Nodes

crazybanane, 2022-05-28:
Hello,

I have the Problem that my Devices list for P2P Devices is full.
Is there a way that you can write an add on or an function that every Node becomes his own Devices list where I can receive more Informations or is there an easier other way?

My RPI works as Standalone without external network and the ESPs are connecting to them.

I Love the RPIEasy Version thanks for this great Work.

enesbcs, 2022-05-29:
Hi! How many units do you have? RPIEasy is able to handle units numbered from 1-254.
If you reached the device limit in RPIEasy its easy to raise, you have to edit rpieGlobaly.py and modify DEVICES_MAX and TASKS_MAX from 96 to 960 for example, if your Raspberry has enough memory.

TD-er, 2022-05-29:
just a question, how compatible is the p2p layer with > 255 tasks?

enesbcs, 2022-05-29:
> just a question, how compatible is the p2p layer with > 255 tasks?
I do not know either. :)

I said "devices" in Devices tasks.
Units can be 1-254 as i wrote.

enesbcs, 2022-05-29:
Sorry, i do not use P2P nowaday, i think i get the problem. Unit task numbers mirrored accross all units, and ESPEasy limits are lower, does not matter what i change on RPIEasy side.

enesbcs, 2022-05-29:
@crazybanane can you tell how many units and how many tasks we are talking currently?

crazybanane, 2022-05-29:
Hi, 
at the moment I use up to 4 Nodes (esp32)
I use on every Node:
1 Device Task for Battery Monitoring
1 up to 4 Task for Analog in Measurements (actually one every Node only 1)

Thanks for Support,👍

enesbcs, 2022-05-29:
The ESPEasy32 can use 32 tasks as i know, so 4x5 is already possible, just make sure, the senders did not use the same Task numbers for sending data.

But I can modify the RPIEasy P2P receiver to assign destination task numbers dynamically, in theory that way the same task number can be used for multiple purposes on the ESP32 side, but it will needs more testing.

enesbcs, 2022-05-29:
Dynamic task number allocation added at commit
https://github.com/enesbcs/rpieasy/commit/6899231bf86ab23e6554335e636b892d9f35b08f

It is disabled by default, if you want to try, than please follow the following order on the RPIEasy receiver node:
1. Disable P2P
2. Delete all remote tasks from devices page
3. Re-enable P2P with "Enable dynamic task number allocation" option checked
4. Force sensorinfo sending on remote side, reboot if needed
5. Wait until remote tasks recreated on RPIEasy receiver node, if the target task is empty, it will try to use, otherwise assign dynamically a new one, based on Sender Unit number and Task number

Device/Task max count raised to 240

crazybanane, 2022-06-18:
Thank you for help will try it

closed 2022-09-18 by enesbcs