enesbcs/rpieasy #254: Max6675 support?

introisback, 2022-06-15:
Hello, how can i use a max6675 thermocoupler on a RPIEasy?
I see at the ESPEasy the max6675 is supported.  
https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P120_Thermocouple.ino

Thx you guys for this project i love it !

TD-er, 2022-06-15:
Maybe better to point to a more recent iteration of the (ESPEasy) code, as this chip is now added to the support of P039, along with other thermocouple chips.
See the [P039 documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P039.html#p039-page)

introisback, 2022-06-15:
I think it will be more to do than the .ino in .py or?

unfortunately i don't know anything about programming and my english skills are lousy. an approximate instruction preferably in german would be helpful. xD

it would definitely be easier to just buy a D1mini.

TD-er, 2022-06-15:
Nope, it surely involves more than renaming a file.
But @enesbcs is more than capable to read C++ and code Python.
My link was more to provide some code examples on how to address this chip you mentioned.

enesbcs, 2022-06-16:
> I think it will be more to do than the .ino in .py or?

Yep, a new .py plugin has to be made by someone, who has a lots of free time.
Basically i would do like this:
1. buy a max6675 for testing
2. grab an rpieasy SPI device code for the skeleton of the new  https://github.com/enesbcs/rpieasy/blob/master/_P210_MCP3008.py
3. use some simple python code to understand how max6675 works, and code a read() function into the new plugin https://github.com/fincham/max6675-spi-python/blob/master/max6675.py
4. test & release

TD-er, 2022-06-16:
@enesbcs Now you're stimulating people to learn how to write code themselves... that can't be the idea behind OpenSource... oh wait... ;)

introisback, 2022-06-16:
I have a lot of free time, but my English is so bad that I can't understand complex texts.
My programming skills are almost 0. 
I know certainly more than 98% of normal people but unfortunately some of the rules (rpieasy) already exceeds my skills - or it does not work all the rules of ESPeasy? 

I look at the 2 links , thank you guys.

TD-er, 2022-06-16:
> I know certainly more than 98% of normal people

Hmm, "normal people" are exceptionally rare, so hard to base statistics on it ;)

introisback, 2022-06-17:
https://youtu.be/T3uc4Jxsx2Y?t=290

https://github.com/tdack/MAX6675

The guy in the video creates a MAX6675.py
-is that helpful in any way?

I hope I don't spam too much. have a nice weekend all of you

enesbcs, 2022-06-17:
> The guy in the video creates a MAX6675.py -is that helpful in any way?

Videos might help if someone needs to disassemble a notebook, but for coding purposes, its not really useful. A test device and the 2 .py file are needed as i wrote earlier.
In the next 6 month it is imaginable that i will have some time to implement it.

TD-er, 2022-06-17:
> In the next 6 month it is imaginable that i will have some time to implement it.

And what will happen in 2023? More or less time available? :)

enesbcs, 2022-07-31:
Support added to P039 plugin at commit https://github.com/enesbcs/rpieasy/commit/a29e53c5d56051f7eeb78eb72ccff423f76aa9d2

enesbcs, 2022-08-01:
> one sensor works perfectly, on occasion I test 2 on one device from if this works without interference I no longer report.

Thank you for testing it!