enesbcs/rpieasy #255: Raspi2 - This hardware has unknown GPIO.

introisback, 2022-06-16:
Hi, is my Raspi2 not supported or is my OS not "good". On my raspi 1b an 3a+ all running fine.

root@solaranzeige:~# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 64.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : 2a01041
Serial          : 00000000ecad0093
Model           : Raspberry Pi 2 Model B Rev 1.1

greetz chris

enesbcs, 2022-06-17:
Raspberry Pi 2 Model B Rev 1.1 (1 GB) Official revision code is: a01041

Your 2a01041 means "Warranty has been voided by overclocking".
https://github.com/raspberrypi/documentation/issues/1888

I will fix autodetection soon.

introisback, 2022-06-17:
I got the Raspi as a gift, thanks for the quick answers, and the great support.

The next coffee and a Max6675 gohn on me.

closed 2022-06-17 by introisback

enesbcs, 2022-06-17:
@introisback fixed in commit https://github.com/enesbcs/rpieasy/commit/56837399474856e6dbe0ce92b98758fa29db4e3e

introisback, 2022-06-17:
Thank you, I hope you have a great weekend. At first glance it looks like it works. 

I will test it later.