enesbcs/rpieasy #256: C016 mysqldb crashes rpieasy

crazybanane, 2022-06-18:
Hey,

Tried to get the DB Storage to work but if I enable it and want to use MYSQL the RPIEasy Programm crashes but raspberry works (available via SSH).
I must reboot it and recover Json files (Delete DB Storage controller in controller.json.

After this it works normally.
Is there a way to get it working and where I can find setting for DB to set save location?

Thanks

enesbcs, 2022-06-18:
> Tried to get the DB Storage to work but if I enable it and want to use MYSQL the RPIEasy Programm crashes but raspberry works (available via SSH). I must reboot it and recover Json files (Delete DB Storage controller in controller.json.

It can happen if mysql client lib crashes for example if your MySQL server daemon is not running at all, or a firewall blocking access. Although in RPIEasy the whole connection process is located inside a try-except loop, so it is very strange. Can you see RPIEasy console logs which may contain serious debugging informations?
You can try Sqlite database, which is saved on the local file system.

> After this it works normally. Is there a way to get it working and where I can find setting for DB to set save location?

I am afraid not really understanding you question. There are no "save location" on MySQL. You can add your mysql servers IP address and port and database name on the controller page. If database did not exist, it will be created... except if connection is unsuccesful.
Please try to reach you mysql server with the correct port, user and pass from the mysql command line client for debugging using SSH.
https://mariadb.com/kb/en/mysql-command-line-client/

If connection succeeds please add more details about mysql database version, server OS and client OS version.

crazybanane, 2022-06-18:
Oh then it's my fault.
I thought that when I use this Function there will automatically made a db where rpieasy will save the data that comes from controller and everything. So when I use SQLite where it will be saved. And what will be saved and how can I chose what will be saved in the DB? Sorry that are noob questions.

enesbcs, 2022-06-18:
> Oh then it's my fault. I thought that when I use this Function there will automatically made a db where rpieasy will save the data that comes from controller and everything. So when I use SQLite [...]

Yep, MySQL client will automatically make the database table, after you installed mysql server of course...

If you select SQLite, it will be one parameter in my memories: the file name with full path, that it will store the data locally.
As it is a controller, you have to make Devices and enable the checkbox at device page to enable them to send Data to the controller, in this case, into the SQLite database.

crazybanane, 2022-06-18:
Jay it works 😁. Nice it can be so easy thank you.

closed 2022-06-18 by crazybanane