enesbcs/rpieasy #257: Timers not working above "15"

Budman1758, 2022-07-05:
I have an irrigation controller and use quite a few different timer events to control a lot of watering events.  I just found out that any timer number 16 and up do not fire events when they expire. Log shows no error when the timer is called but nothing happens when the amount of time expires.  Using timer 15 works as expected and using 16 or above does not.

I was under the impression that there were more available timers than 15. (is timer "0" valid?)  So, either that's wrong or I have found me a bug.  

Was on an earlier version and just updated to newest and behavior was the same.

tonhuisman, 2022-07-05:
[Here](https://github.com/enesbcs/rpieasy/blob/master/rpieGlobals.py#L65) is your answer, I guess 😉, more than double the number of timers available in ESPEasy, but that seems quite easy to fix, though there may be some technical limit, didn't _really_ check how the code is actually working ☺️

Budman1758, 2022-07-05:
That must mean that the first timer number is  0  because  16  does not work.

Budman1758, 2022-07-05:
Well, That was an easy fix, I think.  Changed the max timers to 24 and now numbers 16 and above work.  So far, so good.  

I guess time will tell........   :>)    (pun intended)

Budman1758, 2022-07-11:
Gonna close this. I changed the max number of timers and so far, so good. Feel free to reopen if need be.

closed 2022-07-11 by Budman1758