enesbcs/rpieasy #258: INA219 SLAVE ADDRESS

introisback, 2022-07-31:
Hi in the datasheet of the INA219 says that you can assign 16 different addresses to use 16x INA219 on one I2C.

I wanted to build a 16s BMS with a Raspi, but unfortunately the rpieasy only accepts an INA219 with standard address.

all ina219 with different address are not accepted.

is there a way to fix this or do I have to buy 16 wemos d1 mini to solve the problem?

Thanks in advance for all the answers.
Chris

enesbcs, 2022-07-31:
Based on the Adafruit article it has four option: ["0x40","0x41","0x44","0x45"]
https://learn.adafruit.com/adafruit-ina219-current-sensor-breakout/pinouts

But if you specify the 12 other addresses i can add it to 
https://github.com/enesbcs/rpieasy/blob/master/_P027_INA219.py
at line 81 and line 82 without problem.

I tried to interpret TI database about A0 and A1 addressing, but the hexadecimal values of the binary slave addresses seems very odd to me. For example 0xAF seems incorrect as I2C addresses has 7 bit length which means 0x7F as the largest possible value.

TD-er, 2022-07-31:
> I tried to interpret TI database about A0 and A1 addressing, but the hexadecimal values of the binary slave addresses seems very odd to me. For example 0xAF seems incorrect as I2C addresses has 7 bit length which means 0x7F as the largest possible value.

Some datasheets use the 8-bit notation of the I2C address.
The 8 bits are divided into 7 bit "address" and 1 bit "read/write direction".
NXP also had me confused in some datasheets with this.
If you bitshift the mentioned addresses by 1 to the right, does it then make more sense?

tonhuisman, 2022-07-31:
Some time ago we got a similar request for ESPEasy, so I researched the documentation to enable all 16 advertised I2C addresses. It involves connecting A0/A1 to either GND, VS+ (VCC), SDA or SCL, resulting in this table (from the ESPEasy _P027_INA219.ino source) :
```
      // Many boards, like Adafruit INA219: https://learn.adafruit.com/adafruit-ina219-current-sensor-breakout/assembly
      // A0 and A1 are default connected to GND with 10k pull-down resistor.
      // To select another address, bridge either A0 and/or A1 to set to VS+, SDA or SCL signale.
      //  (0x40) 1000000 (A0=GND, A1=GND)
      //  (0x41) 1000001 (A0=VS+, A1=GND)
      //  (0x44) 1000100 (A0=GND, A1=VS+)
      //  (0x45) 1000101 (A0=VS+, A1=VS+)
      //  (0x42) 1000010 (A0=SDA, A1=GND)
      //  (0x43) 1000011 (A0=SCL, A1=GND)
      //  (0x46) 1000110 (A0=SDA, A1=VS+)
      //  (0x47) 1000111 (A0=SCL, A1=VS+)
      //  (0x48) 1001000 (A0=GND, A1=SDA)
      //  (0x49) 1001001 (A0=VS+, A1=SDA)
      //  (0x4A) 1001010 (A0=SDA, A1=SDA)
      //  (0x4B) 1001011 (A0=SCL, A1=SDA)
      //  (0x4C) 1001100 (A0=GND, A1=SCL)
      //  (0x4D) 1001101 (A0=VS+, A1=SCL)
      //  (0x4E) 1001110 (A0=SDA, A1=SCL)
      //  (0x4F) 1001111 (A0=SCL, A1=SCL)
```
(the hex order seems a bit off, but AFAIR that's how it's documented 😉, it lists the 'easy' addresses first. The device address is determined during power-on, so to change it it needs a power-cycle)

enesbcs, 2022-07-31:
> Some time ago we got a similar request for ESPEasy, so I researched the documentation to enable all 16 advertised I2C addresses. It involves connecting A0/A1 to either GND, VS+ (VCC), SDA or SCL, resulting in this table (from the ESPEasy _P027_INA219.ino source) :

Thank you for this information, it will help a lot! :)

enesbcs, 2022-07-31:
All address added at commit https://github.com/enesbcs/rpieasy/commit/a29e53c5d56051f7eeb78eb72ccff423f76aa9d2

introisback, 2022-07-31:
@ enesbcs   sorry, I thought you knew that the INA219 with up to 16 possible addresses are.

@tonhuisman     thank you exactly what I meant!

-------------------------------------------------------------------

thank you exactly what I meant!
if I use the first 4 addresses at the same time, also the "standard" address no longer works

GND GND 1000000
GND VS+ 1000001
GND SDA 1000010
GND SCL 1000011

also the single modified ones do not work on the espeasy .

If you get this fixed, and I can save the Wemos, get from me in any case a pack of coffee donated. 

best regards

tonhuisman, 2022-07-31:
You may need to remove the pull-down resistor from the board to get it working with the other 12 addresses, as having both a (required) pull-up on SDA/SCL and a pull-down on A0/A1 may prove to be problematic.

introisback, 2022-07-31:
I will give it a try, thanks for your quick replies

@tonhuisman - thanks for the tip, i will have a look at it

there are really super brains behind this project

closed 2022-07-31 by introisback