enesbcs/rpieasy #259: Wifi reconnect after disconnect

Budman1758, 2022-09-12:
Something I have been noticing quite a bit lately with my irrigation controller is the WIFI connection seems to disconnect randomly, and the device stays disconnected. Apparently, no attempt is made to reconnect. I don't know why these disconnects keep happening and the only way to reconnect to the device is to force a reboot by power cycling it as it runs headless.

I would like to see the device reconnect on its own but not sure if this is a "PI" thing or a "RPI" thing. Is there an easy solution to this?

Pi Zero W running Stretch latest RPI version.

enesbcs, 2022-09-13:
Interesting, however network layer is not under RPIEasy control. It is controlled by the Linux kernel. RPIEasy just writes Wifi SSID and password to a config file.
Disconnection may occur when the wifi signal is low or interference is high.
dmesg and syslog may contain information about disconnection reasons. Is it an USB Wifi dongle, or the integrated wifi?

Budman1758, 2022-09-13:
WiFi is the integrated chip on the pi. Any way to use a rule to just restart the WiFi service instead of complete reboot?

enesbcs, 2022-09-13:
wificonnect, wifidisconnect and wifireconnect currently not implemented, but can be in the future through some linux shell scripts

Budman1758, 2022-09-13:
> wificonnect, wifidisconnect and wifireconnect currently not implemented, but can be in the future through some linux shell scripts

Methinks that would be a pretty handy thing.

enesbcs, 2023-08-06:
'wifireconnect' command added to current release, although it may or may not work, as on Linux a lots of things uses network cards beyond the standard ifconfig, ip, iw commands, dhcpcd, and various network managers... this wifireconnect command may work if the integrated power management causes the wifi disconnection.
https://github.com/enesbcs/rpieasy/commit/a7d1219c2e98045c7ac10afb01f529aed55b88b5

The fool-proof solution is a reboot, honestly i am using integrated watchdog on Pi's, as it has a dedicated watchdog hardware.
https://diode.io/raspberry%20pi/running-forever-with-the-raspberry-pi-hardware-watchdog-20202/

enesbcs, 2023-09-26:
Closed as no feedback

closed 2023-09-26 by enesbcs