enesbcs/rpieasy #260: Extend rotary encoder to work with GPIO expanders

Budman1758, 2022-09-17:
Is it possible to expand the available GPIO to use to expander chips such as the MCP23017 or the PFC8474?

I have a unit that's going to be in my truck that needs a bunch of inputs. I need at least 4 rotary encoders and multiple other switches. Extending to other GPIO chips would go a long way to making this possible.

enesbcs, 2022-09-18:
In theory, i can modify Rotary plugin to read existing MCP/PCF plugin states, with Interrupt pin setted up and wired, using the implemented callbacks...

Although while i tested and imlemented Rotary python plugin on my Raspberry Pi, i see that the integrated GPIO access is a bit slow, this results sometimes rotary encoder reading errors, but it mostly works... GPIO access through I2C is much slower than Pi native GPIO access, so i doubt that it will ever work reliably. I guess it needs some testing and a lots of time also.

enesbcs, 2023-12-25:
Encodes definitely needs real-time GPIO access.

I2C encoders may be a viable option for a Raspberry Pi, it will be added sometime.
https://github.com/enesbcs/rpieasy/issues/262

closed 2023-12-25 by enesbcs