enesbcs/rpieasy #262: Feature Request. I2C rotary encoder support.

Budman1758, 2022-10-05:
I stumbled into these at Adafruit the other day and methinks it would be pretty awesome if these could have support. Would make for much simpler setups. Also ability to support many of them over only 2 GPIO.

https://www.adafruit.com/product/4991

enesbcs, 2022-10-06:
It has python example, but as all of it relying on adafruit own libraries it has to be rewritten completely to be used in RPIEasy.
Every rotary has an unique Attiny device integrated on it and all of them needs special cabling? Interesting approach.

Budman1758, 2022-10-06:
Pretty sure the cabling is just standard I2C, power and ground. They use that on a lot of their stuff.

enesbcs, 2022-10-06:
Yep with a non-standard connector

Budman1758, 2022-10-07:
@enesbcs  Would you like me to send you a device and a wiring connector? I have some on the way. Let me know.

enesbcs, 2024-01-01:
@Budman1758 Actually 3GPIO needed, as INT pin is mandatory, but yes 8 I2C rotary encoder could be added. Works pretty fast, although i have tested it only one rotary.
Support added at commit https://github.com/enesbcs/rpieasy/commit/1bbca1375f90db5f690510040d5d86f896ac1d78

enesbcs, 2024-01-01:
Also find a cheaper i2c rotary, which is not made by Adafruit, ordered one of them also for testing:
https://www.duppa.net/shop/i2c-encoder-mini/?v=35b5282113b8

enesbcs, 2024-04-07:
Closed as implemented

closed 2024-04-07 by enesbcs