enesbcs/rpieasy #266: [Feature] Add BME680 Sensor

clumsy-stefan, 2022-12-27:
Would be great to also have the successor of the BME280 included. Unfortunately it doesn't work with the BME280 plugin.

TD-er, 2022-12-27:
There is a plugin for it.
See the extensive docs for it: https://espeasy.readthedocs.io/en/latest/Plugin/P106.html#p106-page
Only we can't use the closed source lib from Bosch to increase accuracy.

clumsy-stefan, 2022-12-27:
Yes, there's a plugin for ESPEasy but not for RPIEasy.... or do I miss something?

TD-er, 2022-12-27:
> Yes, there's a plugin for ESPEasy but not for RPIEasy.... or do I miss something?

Sorry, totally missed the different repo in the URL ;)

enesbcs, 2022-12-27:
Yep there is a bme680 package on PyPI so RPIEasy implemantation wont be complicated.

clumsy-stefan, 2022-12-28:
I'm happy to support and test!

enesbcs, 2023-01-22:
Support added at commit https://github.com/enesbcs/rpieasy/commit/022a368f0612258b651a92ca51aaee80bd6028d6

clumsy-stefan, 2023-01-31:
Great!! I'm currently away, but I'll try to test next weekend! Thanks a lot!

closed 2023-04-01 by enesbcs

clumsy-stefan, 2023-07-02:
Sorry for the late feedback, but finally I got it running (physically) on my RPI... Seemed some voltage issues with the SDA line...
Works smoothly!!
Thanks again for your work!! Really appreciated!!

clumsy-stefan, 2023-07-02:
Actually, I do have an issue with the GAS reading. As per specification it should generate a Index Air Quality (IAQ) reading, which ranges from 0 to 500. The readings I get are all above 50'000 and they increase steadily the shorter the update interval is. So if I click on "run" a number of times the values quickly increase way over 200'000. It seems to be dependant on the update-frequency though.
Not sure if this is a different reading or an issue in the module.
I'll try to investigate, but my python knowledge is still limited...

clumsy-stefan, 2023-07-02:
I included the calculation as per example in the [library](https://github.com/pimoroni/bme680-python/blob/master/examples/indoor-air-quality.py) but obviously with constants for `gas_baseline` and `hum_baseline` which seem to work more or less ok... 
however the issue, that the value seems to increase the more often it's read still persists obviously...