enesbcs/rpieasy #267: Systemd integration

clumsy-stefan, 2023-01-31:
Basically already now RPIEasy can be started via systemd. However currently it runs the "run.sh" script instead of directly the ESPEasy.py (not sure why though, I guess because of the failure-restart loop). This has some downsides (i.e. for monitoring the process). If the if the `rpieasy.service` file is changed to run RPIEasy directly like:
```
[Unit]
Description=rpieasy
After=network.target

[Service]
#ExecStart=/opt/rpieasy/run.sh
ExecStart=/opt/rpieasy/RPIEasy.py
WorkingDirectory=/opt/rpieasy
#StandardOutput=syslog
#StandardError=syslog
SyslogIdentifier=rpieasy
Restart=always
RestartSec=3
User=root

[Install]
WantedBy=multi-user.target
```

the process is directly controlled by Systemd. Also the restart loop and delay can be handled by Systemd.

Just a proposal, IMHO it would be more striaght-forward like this.

enesbcs, 2023-03-30:
Personally I still dont like service, i am using rc.local on every installation...

Yes, run.sh has an infinite loop for safety, and it tries to enter current directory to reach plugins, config and other stuff.
And since you use "ExecStart=/opt/rpieasy/RPIEasy.py" did you see any problem, or is it stable?

clumsy-stefan, 2023-03-30:
Since removing the loop and using systemd to handle restarts etc. I don't have any issues with start/stop/restart anymore. So for me it's stable.

BTW: I use standard Ubuntu 22.10 and not raspbian, probably this also affects how it works..

enesbcs, 2023-08-06:
This modification added to current release
https://github.com/enesbcs/rpieasy/commit/a7d1219c2e98045c7ac10afb01f529aed55b88b5

enesbcs, 2023-09-26:
Closed as no feedback

closed 2023-09-26 by enesbcs