enesbcs/rpieasy #271: Weight - HX711 Load Cell

freddujour, 2023-04-02:
Hi, please add the support for Weight - HX711 Load Cell (https://espeasy.readthedocs.io/en/latest/Plugin/P067.html#p067-typename).
.

enesbcs, 2023-04-15:
I think this code may work with HX711 and python, will se when the test device arrives.
https://github.com/tatobari/hx711py/blob/master/hx711.py

enesbcs, 2023-12-03:
I wired up the 4 sensor to the HX711 but got nothing but random numbers until now. I hate this device very much.

tonhuisman, 2023-12-03:
Reading that HX711 is a bit special, when reading you send a command what data you want to read the _next_ time... Confused the heck out of me when working on the ESPEasy plugin, but finally I got it 🤔 [P067 readHX711() for reference](https://github.com/letscontrolit/ESPEasy/blob/mega/src/src/PluginStructs/P067_data_struct.cpp#L245)

enesbcs, 2023-12-06:
I've added an experimental plugin for this device, but i still unable to get stable readings https://github.com/enesbcs/rpieasy/commit/4b64aa84e0a3b5c644eceb4665ba2774dcb561ea

enesbcs, 2023-12-31:
Tested with a new 10kg sensor, HX711 now produces more stable results... although i am not impressed with this analog device. Closed as added.

closed 2023-12-31 by enesbcs